jneilliii/OctoPrint-PrusaSlicerThumbnails #1: Image not update if file is re-uploaded

TechnoSwiss, 2020-04-21:
Messing around with changing the order of the thumbnails in the gcode and notice that it appears once you have a thumbnail for an object, a new one isn't created when the same filename is re-uploaded.

Other than that seems to be working as expected.

I can't put my hands on the Mini at the moment as it's remote, but my recollection is that the 16x16 thumbnail is used in the object list on the display, and the larger thumbnail is used when viewing or printing the object. I'm not sure if they parse that out, or if they just take the first thumbnail and use it for the object list. Changing the order in the ini does of course change the order in the gcode, but might not work on the Mini, I'll have to test that when I can physically get to the Mini.

As I'm updating my plugins in OctoPrint I'm noticing your name on a majority of the plugins I use, is there some place I can drop you a tip? Same username on Thingiverse, I could use that?

jneilliii, 2020-04-21:
Yeah, I haven't gotten any file removal/checking in there yet at all. I think I have an idea to get around the ordering problem and will be continuing to work on this plugin this week.

My other repositories have my PayPal tip link on them, posted here for convenience.

https://paypal.me/jneilliii

jneilliii, 2020-04-21:
I just pushed an updated version 0.0.2 that now will grab the last thumbnail generated by PrusaSlicer, handles thumbnail overwrite/removal as expected and won't show the inline image/button if a thumbnail is not extracted from the gcode. In order to do some of those changes I had to move the thumbnail url into the file's metadata and as a result they won't show up until after MetadataAnalysisFinished event is fired. This is the same thing that enables the little drop-down arrow button to show the additional data for the file.

jneilliii, 2020-04-21:
I just noticed in the latest version although it does replace the image when re-uploaded, it does not reload the image from the server and in stead shows the browser's cached version. Probably need to add a timestamp token to the end of the url to account for that.

jneilliii, 2020-04-21:
Just forced an update to the 0.0.2 release that handles the caching issue utilizing a timestamp on the url.

TechnoSwiss, 2020-04-21:
That seems to be working very well on my end.

closed 2020-04-22 by jneilliii