jneilliii/OctoPrint-PrusaSlicerThumbnails #10: 0.0.3 problem with The Spaghetti Detective plugin

jmarc67, 2020-04-22:
All was working with 0.0.2 but with the 0.0.3 update no more webcam streaming.

With 0.0.3 stopped, my webcam streaming is back.

I have the Spaguetti Detective plugin

jneilliii, 2020-04-22:
The changes made in 0.0.3 are minimal and related to the back-end server side code and nothing to do with the webcam, so not sure why this would not be working for you. Are there any errors in your octoprint.log?

jmarc67, 2020-04-22:
With 0.0.3 Spaghetti Detective plugin can not load (no problem with 0.0.2)

I have this with 0.0.3 activated:

2020-04-22 20:10:44,287 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 1.622939 seconds: Janus quit! This should not happen. Exit code: 1
2020-04-22 20:10:46,266 - backoff - INFO - Backing off start_gst(...) for 14.7s (Exception: GST failed. Exit code: 1)
2020-04-22 20:10:47,269 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 4.469607 seconds: Janus quit! This should not happen. Exit code: 1
2020-04-22 20:10:54,568 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 8.482731 seconds: Janus quit! This should not happen. Exit code: 1
2020-04-22 20:11:02,063 - backoff - INFO - Backing off start_gst(...) for 20.2s (Exception: GST failed. Exit code: 1)
2020-04-22 20:11:06,235 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 13.722880 seconds: Janus quit! This should not happen. Exit code: 1
2020-04-22 20:11:23,283 - backoff - INFO - Backing off start_gst(...) for 5.0s (Exception: GST failed. Exit code: 1)
2020-04-22 20:11:23,843 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 44.769940 seconds: Janus quit! This should not happen. Exit code: 1
2020-04-22 20:11:29,335 - backoff - ERROR - Giving up start_gst(...) after 8 tries (Exception: GST failed. Exit code: 1)
2020-04-22 20:11:32,361 - werkzeug - INFO - 127.0.0.1 - - [22/Apr/2020 20:11:32] "POST /shutdown HTTP/1.1" 200 -
2020-04-22 20:12:05,188 - octoprint.plugins.thespaghettidetective - ERROR - GST failed. Exit code: 1

No error with 0.0.2

jneilliii, 2020-04-22:
I don't understand what the difference would be here. Maybe @kennethjiang could potentially see what might be the issue here since it looks like an error on his plugin specifically.

jmarc67, 2020-04-22:
No problem jneilliii, your plugin is very nice and very useful. I'll stay with 0.0.2 (i have no subfolder)

I'll try to find what can be this bug tomorrow with more try.

Thanks

kennethjiang, 2020-04-23:
@jmarc67 It looks like gstreamer somehow didn't start successfully. You can turn on debug logging in TSD to see if it'll give more details about this problem: https://www.thespaghettidetective.com/docs/turn-on-debug-logging/.

jmarc67, 2020-04-23:
@kennethjiang  Thanks for your help.
This problem is very strange because 0.0.2 is working very fine and only with the last 0.0.3 i have this bug.
Between 0.0.2 and 0.0.3 you have only 3 differents lines:

self._folder_path = os.path.dirname(path)
if not os.path.exists(self._folder_path):
	os.makedirs(self._folder_path)

i'll do debug-logging ;)

jmarc67, 2020-04-24:
Hello,
Solved ! Reboot ands bad Eth connexion was my own bug. 
Thanks .

closed 2020-04-24 by jmarc67