jneilliii/OctoPrint-PrusaSlicerThumbnails #3: Can't upload gcode with thumbnails.

drewsed, 2020-04-22:
Hi,

thanks for this great plugin.
I am using 
PrusaSlicer Version  2.2.0+w64 and 
Octoprint Version 1.4.0

I created a new Printer Profile in PrusaSlicer, that the `thumbnails =` section appears.
Then I sliced a stl file and uploaded it to Octoprint. It is working fine through PrusaSlicer and drag&drop on the webUI.

Then i added `thumbnails = 16x16,220x124` and from that moment on, the upload to Octoprint fails with the following error:
`Konnte die Datei nicht hochladen. Bitte stelle sicher, dass es sich um eine lesbare valide Datei mit einer dieser Erweiterungen ist: .g, .gco, .gcode
More:

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>`

So, I don't think this is an issue of your plugin, but maybe Octoprint or my configuation.
Any help is appreciated.

Thanks,
Andi

ruralaccent, 2020-04-22:
Same error for me. Cannot upload gcode that has been generated with printer profiles modified as per your instructions. 

[image]

jmarc67, 2020-04-22:
Strange. I do not have the problem. And like you ruralaccent i have the Dashboard plugin.

drewsed, 2020-04-22:
> Strange. I do not have the problem. And like you ruralaccent i have the Dashboard plugin.

Good point: For me I have only this plugin installed.

jmarc67, 2020-04-22:
I have also this plugin: "Upload anything" https://plugins.octoprint.org/plugins/uploadanything/
you could try it perhaps it will help you.

ruralaccent, 2020-04-22:
Hmmm. I had to disable and remove upload anything as it was conflicting with something else and preventing me from being able to print uploaded files. I will try it again, but unless something has changed, then I don't think it'll be a solution for me. If it's down to default Octoprint behaviour blocking upload, then  think this thumbnail plugin should include the code to be able to upload gcode with thumbnails and not rely on the functionality of another plugins (which is a weakness of the dashboard plugin, for example).

ruralaccent, 2020-04-22:
> I have also this plugin: "Upload anything" https://plugins.octoprint.org/plugins/uploadanything/
> you could try it perhaps it would help you.

Just to confirm, I have loaded the Upload Anything plugin and I still cannot upload gcode files with a thumbnail included, made by Prusa Slicer.

drewsed, 2020-04-22:
I installed "Upload Anything" and can upload files via webUI and also the thumbnail is shown correctly, BUT ic cannot upload directly from PrusaSlicer. So this isn't an option for me either.

jmarc67, 2020-04-22:
Ah ok. i didnot try to upload directly from prusaSlicer. Just uploaded files via webUI.
I'll try.

jneilliii, 2020-04-22:
Anyone getting an upgrade error like that needs to include their octoprint.log as it will include more detailed errors for me to find a solution.

ruralaccent, 2020-04-22:
Here are my logs relevant to the upload error...

2020-04-22 11:39:18,717 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 1001.70ms
2020-04-22 11:39:31,010 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 1713.99ms
2020-04-22 11:40:07,624 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 684.03ms
2020-04-22 11:43:19,865 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 813.35ms
2020-04-22 12:06:26,139 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 1068.82ms
2020-04-22 12:51:37,717 - tornado.access - ERROR - 500 POST /api/files/local (::ffff:192.168.0.133) 1704.69ms

ruralaccent, 2020-04-22:
> I installed "Upload Anything" and **can** upload files via webUI and also the thumbnail is shown correctly, **but** i cannot upload directly from PrusaSlicer. So this isn't an option for me either.

Just to confirm, my issue is the opposite to drewsed, because **I can upload thumbnail gcode directly from Prusa Slicer**, but **cannot upload via webUI**.

drewsed, 2020-04-22:
> Just to confirm, my issue is the opposite to drewsed, because **I can upload thumbnail gcode directly from Prusa Slicer**, but **cannot upload via webUI**.

Weired, what did you try first? I firt uploaded from PrusaSlicer and got an error, then I checked uploading via webUI and saw that error you postet as screenshot.

This is my relevant log:

```
2020-04-22 10:38:31,852 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Boot/test.gcode
2020-04-22 10:38:31,853 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/andi/OctoPrint/venv/bin/python2.7 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/andi/.octoprint/uploads/Boot/test.gcode
2020-04-22 10:38:31,852 - octoprint.plugins.prusaslicerthumbnails - INFO - File Added: Boot/test.gcode
2020-04-22 10:38:34,150 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Boot/test.gcode finished, needed 2.30s
2020-04-22 10:42:37,856 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 1953.79ms
2020-04-22 10:44:33,461 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 1208.61ms
2020-04-22 10:58:55,249 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 840.45ms
2020-04-22 10:59:45,272 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 284.49ms
2020-04-22 11:03:36,407 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 474.88ms
2020-04-22 11:09:12,532 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 596.30ms
2020-04-22 12:32:38,594 - tornado.access - WARNING - 409 GET /api/printer?history=true&exclude=state,sd (2003:d5:cf02:c000:cd28:e112:a4f9:36b2) 1.25ms
2020-04-22 14:15:33,389 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 326.52ms
2020-04-22 14:17:09,928 - tornado.access - ERROR - 500 POST /api/files/local (2003:d5:cf02:c000:4535:7d5c:47e2:847d) 504.39ms
```

ruralaccent, 2020-04-22:
Strangley, whilst I can still upload via Prusa Slicer, the thumbnails are no longer showing in Octoprint UI...

2020-04-22 13:47:10,745 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 12.28ms
2020-04-22 13:47:18,155 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 3.78ms
2020-04-22 13:49:19,922 - octoprint.plugins.prusaslicerthumbnails - INFO - File Added: Overglasses_Frame_V5B_0.3mm_PETG_MK3S_58m.gcode
2020-04-22 13:49:26,914 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 7.17ms

[image]

drewsed, 2020-04-22:
@ruralaccent Please use tripple accent signs for proper formatting.

jneilliii, 2020-04-22:
What other plugins do you guys have installed? Moving files could break thumbnails.

drewsed, 2020-04-22:
> What other plugins do you guys have installed? Moving files could break thumbnails.

Only This and the bundled ones, see:

```
2020-04-22 10:30:05,858 - octoprint.plugin.core - INFO - 15 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Logging (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/loginui
|  Plugin Manager (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  PrusaSlicer Thumbnails (0.0.2) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/andi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
```

jneilliii, 2020-04-22:
Can you also share your example gcode files? There's nothing in those logs that give me the detailed errors I was expecting to see.

drewsed, 2020-04-22:
[thumbnail.gcode.txt](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4516263/thumbnail.gcode.txt)
[no_thumbnail.gcode.txt](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4516264/no_thumbnail.gcode.txt)

ruralaccent, 2020-04-22:
> What other plugins do you guys have installed? Moving files could break thumbnails.

I have lots of files and dirs on octoprint, so have to be able to move files. However, it did work fine at first.

jneilliii, 2020-04-22:
> But at this moment the icon does not show up, maybe because its disabled during print:

The thumbnail url gets injected into the metadata upon File Analysis completion (print time estimates, etc.) and becomes available at the same time the "Additional Details" button(drop-down arrow) becomes enabled. This analysis is paused while printing, so the button/inline image will show after your print finishes I hope.

jneilliii, 2020-04-22:
@drewsed looks like your thumbnail gcode works fine. I assume this is related to your multiple instances or because you have an ongoing print.

[image]

drewsed, 2020-04-22:
Well, everything i wrote here was before starting my print.
I can upload a thumbnail.gcode file with plugin "Upload anything" installed, and then the thumbnail was shown fine. But i cannot upload a thumbnail.gcode file from Prusaslicer, so i removed "Upload anything" plugin.

And now after starting my print, i can upload via PS and webUI (without "Upload anything" plugin), maybe because some checks are disabled during a print as you mentioned.
So i think this issue is on Octoprints side because of security checks on the files. I wonder why Prusa Mini users do not complain about this issue all over the internet, when the thumbnail is enabled by default in this printer profile.

> @drewsed looks like your thumbnail gcode works fine. I assume this is related to your multiple instances or because you have an ongoing print.

I do not have multi instances, i just run my webUI on `port 5000 `rather than `80`. But it does not seem to be a problem, because i already have seen a thumbnail once i got a file uploaded.

jneilliii, 2020-04-22:
> I do not have multi instances,

Sorry, I misinterpreted your post on the other issue. Thought you had that as well. Yeah, port number shouldn't effect this at all. I guess the next question is are you uploading to Local Storage or SD card?

drewsed, 2020-04-22:
> Sorry

No problem, glad you clarified :). I'm uploadig to Local Storage.

ruralaccent, 2020-04-22:
I've narrowed it down more on my installation. I can upload fine with a thumbnail to the local root dir, but it fails whenever I try to upload to a subdir. Fails if I used UI and Prusa Slicer upload.

CYberiusPT, 2020-04-22:
Hello.. 

Thanks for this plugin :)

I am having issues as well..

I made the changes in Prusaslicer as mencioned and confirmed the gcode has now the thumbnail code.

Installed the plugin and restarted the service..
I have multiple folders with a few gcodes files of previous prints...
It doesn't show the thumbnail icon as the above.. at first i was thinking that it's because none of that files has the actual thumbnail code as they made before...

After that, unable to upload any file (web or PS). The error was the same as the above...

I deactivated the plugin and it's now the uploading is working fine even with the thumbnail gcode generated..

Activated again, rebooted and the thumbnail icon is  still not showing...

The Displaylayerprogress plugin make changes to the gcode during upload... Maybe is something to do with this?

Not really know what info to give for helping to solve this issue.. 

Thanks!

jneilliii, 2020-04-22:
Are all you guys having this problem uploading to a sub-folder? That may have something to do with it. Also, DisplayLayerProgress may interfere, does the same thing happen with that plugin disabled? Also the thumbnail will not show until after file analysis, so if you use the option to start the print automatically it will not show the button/inline image right away.

ruralaccent, 2020-04-22:
I disabled displaylayerprogress last week for another reason, so defo not that with me.

jneilliii, 2020-04-22:
One last question for now, have to get back to my day job...lol. If you restart OctoPrint are there any errors in the log shown during the initial startup of the application?

drewsed, 2020-04-22:
I have no errors in the whole log except those upload errors i posted above.

ruralaccent, 2020-04-22:
Here's my log from start up...

'''
2020-04-22 16:07:27,582 - octoprint.startup - INFO - ******************************************************************************
2020-04-22 16:07:27,584 - octoprint.startup - INFO - Starting OctoPrint 1.4.0
2020-04-22 16:07:27,584 - octoprint.startup - INFO - ******************************************************************************
2020-04-22 16:07:27,990 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2020-04-22 16:07:28,336 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2020-04-22 16:07:28,363 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-04-22 16:07:28,791 - octoprint.plugin.core - INFO - Plugin Anonymous Usage Tracking is disabled.
2020-04-22 16:07:29,136 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py:5: ExtDeprecationWarning: Importing flask.ext.babel is deprecated, use flask_babel instead.
  from flask.ext.babel import gettext

2020-04-22 16:07:29,138 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py:6: ExtDeprecationWarning: Importing flask.ext.login is deprecated, use flask_login instead.
  from flask.ext.login import current_user

2020-04-22 16:07:29,711 - octoprint.plugin.core - INFO - Plugin Prusa Leveling Guide (1.0.7) is disabled.
2020-04-22 16:07:30,191 - octoprint.plugin.core - INFO - Plugin DisplayLayerProgress Plugin (1.20.0) is disabled.
2020-04-22 16:07:30,225 - octoprint.startup - INFO - Added settings overlay from plugin dashboard
2020-04-22 16:07:30,227 - octoprint.plugin.core - INFO - Found 42 plugin(s) providing 39 mixin implementations, 73 hook handlers
2020-04-22 16:07:30,477 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-04-22 16:07:30,491 - octoprint.server - INFO - Intermediary server started
2020-04-22 16:07:30,492 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-04-22 16:07:30,590 - octoprint.plugin.core - INFO - Found 42 plugin(s) providing 39 mixin implementations, 73 hook handlers
2020-04-22 16:07:30,609 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2020-04-22 16:07:31,823 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2020-04-22 16:07:31,826 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2020-04-22 16:07:31,827 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2020-04-22 16:07:31,828 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plugin_announcements_read')")
2020-04-22 16:07:31,829 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2020-04-22 16:07:31,829 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2020-04-22 16:07:31,830 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2020-04-22 16:07:31,831 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2020-04-22 16:07:31,832 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-04-22 16:07:31,833 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-04-22 16:07:31,834 - octoprint.server - INFO - Added new permission from plugin printer_safety_check: PLUGIN_PRINTER_SAFETY_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_printer_safety_check_display')")
2020-04-22 16:07:31,835 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: u"Need(method='role', value=u'plugin_softwareupdate_check')")
2020-04-22 16:07:31,835 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: u"Need(method='role', value=u'plugin_softwareupdate_update')")
2020-04-22 16:07:31,846 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2020-04-22 16:07:31,858 - octoprint.plugins.cancelobject - INFO - No beforegcode defined
2020-04-22 16:07:31,859 - octoprint.plugins.cancelobject - INFO - No aftergcode defined
[... cut ...]
2020-04-22 16:08:04,909 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.133
2020-04-22 16:08:11,547 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.133
2020-04-22 16:08:11,760 - octoprint.server.util.flask - INFO - Passively logging in user ******* from ::ffff:192.168.0.133
2020-04-22 16:08:11,761 - octoprint.access.users - INFO - Logged in user: *******
2020-04-22 16:08:15,030 - octoprint.server.util.sockjs - INFO - User ******* logged in on the socket from client ::ffff:192.168.0.133
'''

jneilliii, 2020-04-22:
@ruralaccent is you disable PrintTimeGenius Plugin does the issue remain?

ruralaccent, 2020-04-22:
> @ruralaccent is you disable PrintTimeGenius Plugin does the issue remain?

Just tried it. Makes no difference, unfortunately.

jneilliii, 2020-04-22:
I have successfully re-created the same error by trying to upload to a sub-folder, so at least that might be related to your problem @ruralaccent and @drewsed I will look into why that is happening and get it resolved as soon as possible.

jneilliii, 2020-04-22:
Alright, I think I have the issue fixed, at least with my testing. If you guys don't mind trying, install an updated version by using the URL below in plugin manager and give it a go.

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/0.0.3.zip

safusu, 2020-04-22:
Just tested v0.0.3, not getting the 500 internal server error anymore

ruralaccent, 2020-04-22:
Yes. v0.0.3 seems to work for me know. Many thanks :)

jneilliii, 2020-04-22:
Thanks @safusu, is the thumbnail showing or are you currently printing and the thumbnail url hasn't been updated yet due to file analysis not being completed?

closed 2020-04-22 by jneilliii

safusu, 2020-04-22:
> Thanks @safusu, is the thumbnail showing or are you currently printing and the thumbnail url hasn't been updated yet due to file analysis not being completed?

Thumbnail shows properly 
[image]

jneilliii, 2020-04-22:
Great, I just officially released an update with these changes. You guys don't need to do anything that have already manually updated.

ruralaccent, 2020-04-22:
Thanks for this great plugin and for fixing it so quickly :)

drewsed, 2020-04-22:
Yes, thanks so much. I think it's not a surprise anymore, now it is working for me fine, too.

Great :-*

PR #8 (2020-04-22): 0.0.3
Fixed issue related to uploading to sub-folders, resolves #3