jneilliii/OctoPrint-PrusaSlicerThumbnails #4: Incorrect url for multiinstance config

ochm, 2020-04-22:
Hi there,

I'm running octoprint on 2 instances on one RPi.
http://192.168.73.245/P2
http://192.168.73.245/P1
defined in haproxy config
[haproxy.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4515095/haproxy.zip)

But this plugin generates image url
http://192.168.73.245/plugin/prusaslicerthumbnails/thumbnail/Shape-Box_0.6n_0.1mm_PETG.png?20200422110821
As you can see, the part which defines which Octoprint instance is called (/P1 or /P2) is missing in your image url.

Please, can you repair it?

jneilliii, 2020-04-22:
Sorry, didn't anticipate multi instance setups. I'll look into it.

drewsed, 2020-04-22:
Are single instances with different port than `80` also affected by this?

jneilliii, 2020-04-22:
The URL that is used shouldn't be effected by this as it's a relative URL to the base server instance. For example, in the image link above it's actually using `/plugin/prusaslicerthumbnails/thumbnail/Shape-Box_0.6n_0.1mm_PETG.png?20200422110821` not the full URL. It might be possible to eliminate the first `/` to resolve this issue. If you'd like to test that theory, if you can access your metadata.json file with a text editor and adjust the thumbnail field to remove that first `/` it would go a long way in helping me resolve this issue. I don't have multi-instance installation so can't really debug this myself.

ochm, 2020-04-22:
Yes! 
Remove leading / and restart of Octoprint helped!
Now the  image is displayed (for printer1 is url http://192.168.73.245/P1/plugin/prusaslicerthumbnails/thumbnail/kvetinacek2-0.6-PLASTIA_0.6n_0.4mm_PETG.png?20200422220558
In the metadata file is:
        "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/kvetinacek2-0.6-PLASTIA_0.6n_0.4mm_PETG.png?20200422220558",

So it looks like easy repair for you. :)

jneilliii, 2020-04-22:
Awesome, thanks for the testing. I'll add that to my list of changes for the next version update.

ochm, 2020-04-22:
You are welcome, maybe small amount on PayPal will help you too. :) Have a nice day.

ochm, 2020-04-24:
I made upgrade, but the behavior is still the same. filename begins with /

jneilliii, 2020-04-24:
New version released that resolved this issue.

closed 2020-04-24 by jneilliii

ochm, 2020-04-24:
Perfect, works OK. Thank you.

PR #15 (2020-04-24): 0.0.4
Moved URL injection of file metadata to FileAdded event, resolves #13, resolves #7 
Removed leading / from URL to resolve multi-instance related issues, resolves #4
Changed alignment option to a select list, resolves #11 
Improved performance using regex pattern matching instead of line processing
Deleting folder will now also delete extracted thumbnails