jneilliii/OctoPrint-PrusaSlicerThumbnails #5: Image preview not available...

dirkvranckaert, 2020-04-22:
So I installed the plugin manually (not available through the plugin manager), everything went fine, I have the configuration options in the settings.

I also modified my printer configuration file as described in the settings and export a GCODE with prusaslicer. But no image icon is showing up in the list... I tried enabling the preview option but that does not change a thing either.

jmarc67, 2020-04-22:
I had the same problem as you dirkvrankaert, but now it's working fine.
I had to refresh file list, to see image icon.

Nice plugin

dirkvranckaert, 2020-04-22:
Ah I did all kinds of restarts and refreshes since this morning but no luck. And now (after a successful print) it suddenly there... :-) Great plugin!

closed 2020-04-22 by dirkvranckaert