jneilliii/OctoPrint-PrusaSlicerThumbnails #6: Plugin with Themeifiy

flc85, 2020-04-22:
Hello,
Sorry I don't understand what I must to do in Themeify with this line: 
`div.row-fluid.inline_prusa_thumbnail > img`
Can you explain me?

jneilliii, 2020-04-22:
So the sizing option was added to the plugin's settings, so that is kind of an old reference that I need to probably remove. What I meant to say is that the file list itself may need to be resized to show more items in the list when the inline image option is enabled. For that you would enter the following configuration in themeify.

| Selector                                            | CSS_Rule   | Value            |
|-----------------------------------------------------|------------|------------------|
| #files > div > div.gcode_files > div.scroll-wrapper | min-height | 800px !important |

thisiskeithb, 2020-04-22:
In addition to adding the custom CSS mentioned above under the Themify plugin & modifying my PrusaSlicer configs, I didn't realize I had to enable "Use inline thumbnails." under the "PrusaSlicer Thumbnails" options.

After that, magic!

[image]

closed 2020-04-22 by jneilliii

flc85, 2020-04-22:
Thanks for all