jneilliii/OctoPrint-PrusaSlicerThumbnails #7: Thumbnails Broken With File Moves

jneilliii, 2020-04-22:
Strangley, whilst I can still upload via Prusa Slicer, the thumbnails are no longer showing in Octoprint UI...
```
2020-04-22 13:47:10,745 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 12.28ms
2020-04-22 13:47:18,155 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 3.78ms
2020-04-22 13:49:19,922 - octoprint.plugins.prusaslicerthumbnails - INFO - File Added: Overglasses_Frame_V5B_0.3mm_PETG_MK3S_58m.gcode
2020-04-22 13:49:26,914 - tornado.access - WARNING - 404 GET /plugin/prusaslicerthumbnails/thumbnail/Overglasses_Frame_V3b2_(2no)_0.3mm_PETG_MK3S_1h44m.png?20200422134115 (::ffff:192.168.0.133) 7.17ms
```
[image]

I move the files from root to a different directory. Thumbnail was fine at first, but then I restarted the server and now I don't see the thumbnail, just a placeholder. 

Edit: I did restart Prusa Slicer, so that's not the issue. :)

_Originally posted by @ruralaccent in https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/issues/3#issuecomment-617761075_

jneilliii, 2020-04-23:
@ruralaccent, how are you moving your files? I'm looking into this now while doing some other fixes and wanted to see how much was going to be involved in resolving this.

ruralaccent, 2020-04-24:
I use the file manager plugin.

PR #15 (2020-04-24): 0.0.4
Moved URL injection of file metadata to FileAdded event, resolves #13, resolves #7 
Removed leading / from URL to resolve multi-instance related issues, resolves #4
Changed alignment option to a select list, resolves #11 
Improved performance using regex pattern matching instead of line processing
Deleting folder will now also delete extracted thumbnails
closed 2020-04-24 by jneilliii