jneilliii/OctoPrint-PrusaSlicerThumbnails #13: Preview icon is not available when choosing 'Start printing after upload' in PrusaSlicer

tychop, 2020-04-23:
Tiny bug I think:
The preview icon is not available when choosing 'Start printing after upload' in PrusaSlicer.
It only shows up when you do not print it immediately, or when the print is finished.

For the rest, 👍🏻💪🏻
Great job.

jneilliii, 2020-04-23:
Not really a bug, it was actually programmed that way. The image itself is extracted during the upload process, but because file analysis doesn't start/ complete before the print starts the url doesn't get added to the metadata until after print completion. This was a limitation in the core octoprint plugin framework because none of the metadata exists until after that file analysis is completed. It's the same reason why the little arrow drop down doesn't become enabled until after print completion with auto start print as well. 

I will verify this with the plugin framework just to be sure.

tychop, 2020-04-23:
Ah, yes. That explains a lot. Nowhere to hook into to run your code... Thnx

bbrendon, 2020-04-23:
Ahhh... Can this be added to the readme? I think a lot of people are going to wonder about this.

jneilliii, 2020-04-23:
That is probably a good idea @bbrendon, but I was hoping to be able to program around it, just haven't had the time to look into it yet.

jneilliii, 2020-04-23:
Updated this repositories readme with a note that I'm working on it for the next release.

jneilliii, 2020-04-24:
New version released that resolved this issue.

closed 2020-04-24 by jneilliii

tychop, 2020-04-24:
Works !!!
Thnx

PR #15 (2020-04-24): 0.0.4
Moved URL injection of file metadata to FileAdded event, resolves #13, resolves #7 
Removed leading / from URL to resolve multi-instance related issues, resolves #4
Changed alignment option to a select list, resolves #11 
Improved performance using regex pattern matching instead of line processing
Deleting folder will now also delete extracted thumbnails