jneilliii/OctoPrint-PrusaSlicerThumbnails #14: PrusaSlicer 2.0.0-alpha4 does not start after modifiyng printer settings

Markowia, 2020-04-23:
I use this version of PrusaSlicer (in the title) and it denies to start after i add the thumbnail sizes in the printer configuration.
Is this a problem with the slicer version?
Which version is safe?

jneilliii, 2020-04-23:
The feature was added in a newer release of PrusaSlicer, just not sure specifically. I'm using 2.2.0+win64, which is the latest stable release version.

closed 2020-04-23 by jneilliii

Markowia, 2020-04-24:
> The feature was added in a newer release of PrusaSlicer, just not sure which version specifically. I'm using 2.2.0+win64, which is the latest stable release version I believe.

You are right, the stable version works flawless. Just the alphas are not good handeling this profile change. Thank you very much for the assistance and I hope I didn't waste your time with this.

jneilliii, 2020-04-24:
no problem at all.