jneilliii/OctoPrint-PrusaSlicerThumbnails #16: "Gcode editor plugin" wont display

TheNore, 2020-04-23:
am using the gcode editor plugin which inserts an icon next to the "open" and "print" icons below the gcode file in the file manager. The thumbnail will not display because of this. Once disabled the thumbnail displays fine. Inline, and with button.

jneilliii, 2020-04-24:
Does it only effect the button, or inline as well? More than likely it's because we are using a similar approach to inject the buttons into the default UI.

TheNore, 2020-04-24:
Yes, when the the gcode editor is disabled inline then shows up.
It does not show up when the gcode editor is enabled.

jneilliii, 2020-04-24:
I think you might have not enabled the inline option in settings? 

[image]

When that option is not enabled you will see a button to open the pop-up like this.

[image]

I just installed the GcodeEditor plugin and it's working for me.

TheNore, 2020-04-24:
It works now, for some reason I had to disable and re enable the gcode
plugin. thanks for clarifying

closed 2020-04-24 by jneilliii