jneilliii/OctoPrint-PrusaSlicerThumbnails #18: 100% CPU and Lock with the version 0.0.4

dianlight, 2020-04-24:
Hi with the last version 0.0.3 if I send a job to Octoprint all lock. And the Octoprint take 100CPU.
Disable the plugin or use the previus version (0.0.3) all work great.

No specific log found. 
You can download an example of the gcode from url  [Gcode File. Link valid until 1/5/2020](https://www.dropbox.com/t/IXg47Wa4kamAHDNS)

jneilliii, 2020-04-24:
How are you sending it, via PrusaSlicer directly or downloading and uploading to OctoPrint?

jneilliii, 2020-04-24:
Just tested your file and it works fine on my pi.

[image]

Please provide the following information.

1. What other plugins do you have installed?
2. Are there any errors in octoprint.log since installing the update?

dianlight, 2020-04-24:
No error on log file. Simply 100% CPU and Octroprint lock until service restart.
I'm on a Rpi 3B+ (OctoPrint 1.4.0 running on OctoPi 0.17.0) 

My plugin outside the standard are (a lot! ;)):

Action Command Prompt Support
Auto Terminal Input (0.1.0)
Cancel Objects (0.4.1)
CommandSplitter (0.1.0)
Dashboard (1.11.4)
DeleteAfterPrint Plugin (1.7.0)
DisplayLayerProgress Plugin (1.20.0)
Dragon Order (0.1.4)
FileManager (0.1.4)
Metadata Preprocessor (0.1.0)
Octolapse (0.3.4)
OctoPod Plugin (0.2.5)
PrintTimeGenius Plugin (2.2.1)
PSU Control (0.1.9)
Python 3 Check (0.1.3)
Resource Monitor (0.2.2)
Tab Order (0.5.6)
Terminal Commands Extended (0.1.4)
The Spaghetti Detective (1.1.5)
Themeify (1.2.2)
Ultimaker Format Package (0.1.4)

I will try to disable all and retry.... 

L.

jneilliii, 2020-04-24:
>Metadata Preprocessor (0.1.0)

I would start with this one.

dianlight, 2020-04-24:
I found the plugin! ;)
Is CommandSplitter (0.1.0). If I disable it and restart OctoPrint all works. 
I think it touch the Thumbnail comment and your code go in loop.

I open the Issue on the CommandSplitter repository because it is wrong and bad but I suggest to you to check what is the infinite loop cause to prevent future issue.

Regards
L.

jneilliii, 2020-04-24:
I installed CommandSplitter on my pi and was still able to upload your gcode files without issues.

foosel, 2020-04-24:
I also can't see how CommandSplitter could cause issues here since it doesn't touch anything in comments. 

Also, I'm pretty sure this is a case of a language barrier, but:

> because it is wrong and bad

^ this phrasing could really use some polish.

dianlight, 2020-04-24:
I can't help you much. What I can say is that I have all the plugins disabled.
The only oddity is that if I disable CommandSplitter I am not asked to restart Octoprint and without restart the lock return.

Test 1 --> LOCK
- OctoPrint-PrusaSlicerThumbnails ENABLED
- OctoPrint-CommandSplitter ENABLED

Test 2 --> LOCK
- OctoPrint-PrusaSlicerThumbnails ENABLED
- OctoPrint-CommandSplitter **DISABLED with no RESTART**

Test 3 --> NO LOCK
- OctoPrint-PrusaSlicerThumbnails ENABLED
- OctoPrint-CommandSplitter **DISABLED and RESTART**

Test 4 --> NO LOCK
- OctoPrint-PrusaSlicerThumbnails DISBALED
- OctoPrint-CommandSplitter ENABLED

If it helps, I can try an installation from scratch. But it takes me some time.

> > because it is wrong and bad
> ^ this phrasing could really use some polish.

sorry but english is not my native language. It's a way of saying.

jneilliii, 2020-04-24:
So @foosel, there is something strange here that I think might be a timing issue. Although I was able to initially upload the gcode and the image is extracted when CommandSplitter removes the empty lines it causes this plugin to not grab the last thumbnail, but the first one for some reason. I have not been able to reproduce 100% CPU load at all. 

The reason I say that is because the following plugins when combine might be fighting each other?

Cancel Objects (0.4.1)
CommandSplitter (0.1.0)
DisplayLayerProgress Plugin (1.20.0)
Metadata Preprocessor (0.1.0)
PrintTimeGenius Plugin (2.2.1)
PrusaSlicerThumbnails (0.0.4) - uses FileAdded event to read the gcode file and set_additional_metadata..

foosel, 2020-04-24:
I could see it being an issue of timing maybe, that would be a bug in OctoPrint though and not in the plugin. I'm unsure right now at which point the FileAdded event fires, if it fires before the file has been modified (CommandSplitter is a preprocessor, not an event handler) that could be an issue due to concurrent modification. Would it possible for you to investigate this a bit more? I'm still drowning in work here.

In general the event handler plugins should be called one after the other, so there should not be any infighting.

jneilliii, 2020-04-24:
Sure, no problem Gina. I initially had the image extraction as a pre-processor, but since I'm not actually modifying the data just pulling it out I took that stuff out in favor of the event callback. If I do figure out it is some kind of timing issue between the event and processor I'll post an issue.

dianlight, 2020-04-24:
I report here the analysis made and the data on the  [Issue #2 on OctoPrint-CommandSplitter](https://github.com/OctoPrint/OctoPrint-CommandSplitter/issues/2) so you can have more information.

The CommandSplitter plugin seems to remove the empty lines between the thumbails. I think that's exactly the problem.

[image]

[Pin_0.3_2_Test.gcode --> Original GCODE ](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4529910/Pin_0.3_2_Test.gcode.txt)
[Pin_0.3_2_Test.ReDownloaded.gcode --> GCODE redownloaded from Octopi](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4529912/Pin_0.3_2_Test.ReDownloaded.gcode.txt)

jneilliii, 2020-05-22:
@dianlight I think I may have just though of a way to deal with this. If I were to do a replace of
```
;
;
```
with 
```
;

;
```
I think it might resolve this bug. Please try the attached zip file for me and see if that works. Just use the browse button in the plugin manager and select the zip file and restart octoprint/reload.

[OctoPrint-PrusaSlicerThumbnails.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4670599/OctoPrint-PrusaSlicerThumbnails.zip)

jneilliii, 2020-05-24:
Closing as I think this is resolved now. Feel free to re-open if that's not the case.

closed 2020-05-24 by jneilliii