jneilliii/OctoPrint-PrusaSlicerThumbnails #19: Add thumbnail to State area

soerup, 2020-04-24:
[image]

Is it possible to also add thumbnail here?

jneilliii, 2020-04-24:
That might be possible, I'll just have to check to see if the same bindings are available in that panel that are in the file list panel.

jneilliii, 2020-04-24:
Would it make sense that when you select a file or start the print of the file that it hides the inline thumbnail from the file list or just leave it as is? Or potentially only inject the thumbnail into the state panel when inline images are not enabled? Make all these variations optional in settings?

closed 2020-04-25 by jneilliii

jneilliii, 2020-04-25:
@soerup, new version 0.0.5 released that incorporates this as an option, enabled by default.

PR #21 (2020-04-25): 0.0.5
modifications to allow both UltimakerFormatPackage and this plugin to play nice together
added note to restart PrusaSlicer once the ini file has been updated, resolves #20 
add state panel thumbnail for selected file if available, resolves #19