jneilliii/OctoPrint-PrusaSlicerThumbnails #20: Not getting a Thumbnail button in the file list

PRGeno, 2020-04-24:
The plugin seems to be loaded in my OctoPrint 1.4.0 instance The plugin configuration page does show up in the OctoPrint settings.

However, I don't get the new button showing up in the file list as expected.  

I have a GCode file with the thumbnail embedded in the file list, but I would think the button would show up regardless of what was in the file.

Any idea what I might be missing?
[image]

jneilliii, 2020-04-24:
So not sure if this is the issue or not, but only gcode files uploaded after installing the plugin will show the inline image (if enabled) or button. That being said, it looks like you are running a custom theme, that could potentially be breaking something if the above is not the case. Would need more details from you on your setup, like what plugins are installed, etc.

PRGeno, 2020-04-24:
I do have a number of plugins loaded including a few that affect the Sidebar.  I have Sidebar Temp Graph to add the temperature graph to the sidebar, used Dragon Order to change the order of the sidebar, and used Themeify to change the colors.

I did load your plugin first and then added the gcode for with the thumbnail embedded, but no button showed up.  I tried with the inline option both disabled and enabled.

I can provide more information about my two setups if you like.

It would be a shame if any of these other plugins caused your plugin to not work.  The other plugins are more necessary for my user than thumbnails.

jneilliii, 2020-04-25:
I'll definitely need more information if you want some assistance in figuring out why it's not loading. Can you provide an example gcode file, check your octoprint.log for errors, and the browser's developer console (F12) for errors on initial loading of the webpage.

PRGeno, 2020-04-25:
I'll find some time to do more testing this weekend. I'll get back to you.

chilternflyer, 2020-04-25:
PRGeno: I had the same issue. I'm running the same octoprint plugins/ version / printer that you are. 
I found that  after adding "thumbnails = 16x16,220x124" to my printer ini file it still didn't work. 

Check that your GCode files actually have a thumbnail in them. I think you'll find that you have to exit Prusa Slicer completely and restart it before it adds thumbnails after you change the config!

My issue wasn't the plug-in :-(

jneilliii, 2020-04-25:
Good point @chilternflyer,

chilternflyer, 2020-04-25:
jneilliii: - Since I've got it working it has already become my favorite plugin. :-)

PRGeno, 2020-04-25:
> PRGeno: I had the same issue. I'm running the same octoprint plugins/ version / printer that you are.
> I found that after adding "thumbnails = 16x16,220x124" to my printer ini file it still didn't [...]

Well that was certainly an easy fix.  I tried to get it working in one sitting, never thinking of restarting Prusa Slicer in the process.  I figured PS read the Printer Settings file every time the printer was switched.  Apparently not.

chilternflyer: Thanks for chiming in with the solution

jneilliii: Thanks for the cool plugin.  I guess this issue can be put into the troubleshooting tips and tricks.

closed 2020-04-25 by PRGeno

jneilliii, 2020-04-25:
Thanks guys, I'll add that to the readme.

PR #21 (2020-04-25): 0.0.5
modifications to allow both UltimakerFormatPackage and this plugin to play nice together
added note to restart PrusaSlicer once the ini file has been updated, resolves #20 
add state panel thumbnail for selected file if available, resolves #19