jneilliii/OctoPrint-PrusaSlicerThumbnails #24: PrusaSlicer Crashing when loading edited profile

steve220, 2020-04-26:
Unfortunately, when I create a copy of the profile (for the MK3S) and add the thumbnail line to it, Prusaslicer crashes when I try to load the new printer profile. I'm using PS 2.1.1 64bit and edited the ini file with notepad++. I've uploaded the resulting ini file in case anyone wants to take a look and provide any suggestions.

[Original Prusa i3 MK3S - Smith.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4535587/Original.Prusa.i3.MK3S.-.Smith.zip)

jneilliii, 2020-04-26:
If the `thumbnails` variable was missing completely the issue might be the version of PrusaSlicer you are using. I'm not sure when this feature was added to that application, but I'm running version 2.2.0+win64.

rendez2k, 2020-04-26:
So I seem to be having a similar issue. The plugin works fine as I setup a Mini profile in PrusaSlicer 2.2.0 and the thumbnails work great. 

For my normal MK3s + MMU2s, I went to the config folder with my duped profile (C:\Users\rende\AppData\Roaming\PrusaSlicer\print\0.20mm QUALITY @MK3 -+ Thumb.ini) and edited the file. 

As above, no thumbnails line in there and adding it seems to make no difference. Thumbnails are not present in the exported gcode. 

[0.20mm QUALITY @MK3 -+ Thumb.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4536530/0.20mm.QUALITY.%40MK3.-%2B.Thumb.zip)

jneilliii, 2020-04-26:
I'm not sure if this is the issue, but that looks like a filament or other config. The path for the ini should be `C:\Users\rende\AppData\Roaming\PrusaSlicer\printers\`

rendez2k, 2020-04-26:
So my folder was 'printer' and not 'printers' for some reason but that worked! I did find a config file in there with the thumbnail line. Many thanks!

jneilliii, 2020-04-26:
Cool, I was going from memory.

closed 2020-04-26 by jneilliii

steve220, 2020-04-27:
Updating my version of prusaslicer fixed this.