jneilliii/OctoPrint-PrusaSlicerThumbnails #25: State thumbnail visibility

pooyashahidi, 2020-04-26:
If the user choose to have the thumbnail in "State" panel while printing and the gcode is located in a folder, the thumbnail disappears when user navigate out of that folder or to another folder
in this example I'm printing a file from "Speaker_Box" folder and when I'm viewing the root folder there is no thumbnail:
[image]
</br>
 but if I navigate to the mentioned folder it appears:
[image]

jneilliii, 2020-04-26:
Thanks for this. I think I know why it does that. If you switch back to the folder the file is in does the thumbnail show back up?

pooyashahidi, 2020-04-26:
yes It shows up when you are in the same folder

jneilliii, 2020-04-29:
This should be fixed in version 0.0.7 that was just released.

PR #26 (2020-04-29): 0.0.7
added a file crawler in settings that will extract missing thumbnails, related to #22 
switched state panel thumbnail logic, resolves #25
closed 2020-04-29 by jneilliii