jneilliii/OctoPrint-PrusaSlicerThumbnails #27: Touch UI Interface Icons

TheNore, 2020-04-28:
After trying to manipulate the different view options i am unable to get the files to print from my touch UI instance. Only the website works. which is very inconvenient, and breaks my workflow.

This is a very important feature and it renders the use of touch ui semi useless as a standalone controller.

Images below reflect inline and stateless views. Paul de Vries (BillyBlaze) Might have to alter Touch UI to work with this plugin. 
[image]

[image]

jneilliii, 2020-04-28:
It looks like you also have the gcode editor plugin installed? I personally don't use TouchUI in preference of OctoDash, but have made changes to other plugins to get around TouchUI issues by disabling it completely if TouchUI is running and enabled.

jneilliii, 2020-04-28:
And maybe you should also cross-post an issue on TouchUI's repository so @BillyBlaze actually sees it.

jneilliii, 2021-04-17:
I cannot re-create this issue with gcode editor and touchui and this plugin all installed at the same time. So I think this was either fixed by those other plugins, are some of the changes in my current 0.1.5rc6 version has fixed this. Closing for now, please comment if this is still an issue for you with 0.1.5rc6 version of the plugin.

closed 2021-04-17 by jneilliii