jneilliii/OctoPrint-PrusaSlicerThumbnails #28: Missing image button

fornellas, 2020-05-04:
[image]

I installed PrusaSlicer Thumbnails 0.0.7 with Octoprint 1.4.0 and Python 3.6.9, but unfortunately, there's no new image button. I double checked, this gcode **does** have the thumbnail comments at the first few lines. I did not spot any errors on `octoprint.log` as well.

jneilliii, 2020-05-04:
Was the gcode uploaded before or after the plugin was installed?

fornellas, 2020-05-04:
After.

jneilliii, 2020-05-04:
Please download the gcode from octoprint and share here.

apphead, 2020-05-04:
Same here.  PrusaSlicer Thumbnails 0.0.7 with Octoprint 1.4.0 and Python 2.7.16. 
Thumbnail is displayed in the list though - configured in Themefy as per your instructions...

[image]

Here's the gcode:

[Test Titan Aero hole spacing_0.2mm_PLA_MK3S_16m.txt](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4577393/Test.Titan.Aero.hole.spacing_0.2mm_PLA_MK3S_16m.txt)

jneilliii, 2020-05-04:
@apphead, different "issue" the thumbnail is displayed inline because you have the inline option enabled in the plugin's settings. @fornellas says he doesn't see either option the button or the inline image. It also appears he's running Python 3, which could be the underlying problem.

jneilliii, 2020-05-04:
I can confirm this is a python 3 bug @fornellas, I'm looking into it now.

apphead, 2020-05-04:
@jneilliii I was trying to demonstrate that the gcode is not the issue as the thumb gets displayed in the list. But I also miss the image button (updated image above to reflect that). 
Strangely, I get the following warning in the settings after a scan:
[image]

jneilliii, 2020-05-04:
I've seen this before with the gcode editor plugin, which apparently you have installed.

jneilliii, 2020-05-04:
I think I've got this fixed for you @fornellas, just need to merge in the changes and push a release update.

closed 2020-05-04 by jneilliii

apphead, 2020-05-04:
You are right about the gcode editor plugin!
I disabled it and rebooted octoprint -> button still missing
I uninstalled and rebooted octoprint -> missing
I re-scanned for thumbs -> missing (but warning disappeared)
I rebooted RPi -> missing
Re-installed gcode editor and rebooted -> missing
Me -> no clue (but still happy with inline thumb

jneilliii, 2020-05-04:
@apphead, did you uncheck the inline image option in the plugin's settings when you disabled the gcode editor plugin? As mentioned before, the button will not be there if that option is enabled. You can click the inline thumbnail to open the larger view BTW.

apphead, 2020-05-05:
Hahaha, turns out that the gcode editor is not the culprit. It's either inline thumb or button. How the f"ç* could I miss that. Thanks for pointing it out! All good now

fornellas, 2020-05-05:
@jneilliii thanks for looking into this.

I manually updated with the zip file (as it seem 0.0.8 is not published yet). I re-scanned the files, and:

[image]

\o/

jneilliii, 2020-05-05:
It is published, in those situations you have to go to advanced and force a refresh because the releases are cached.