jneilliii/OctoPrint-PrusaSlicerThumbnails #30: Running Octoprint in dual setup...

Snille, 2020-05-05:
Hi! I'm running multiple instances of Octoprint on the same system to be able to control two printers from the same RPi. I followed [this guide](https://www.youtube.com/watch?v=7Saa1HpLRoM) and all works well. However, for some reason I only see the thumbnail pictures on one of the instances (the one I uploaded the gcode to), So if I upload to my "MK2.5", I only see the thumb there, and a "broken" thumb in the "MK3" instance. I'm using the "same" upload directory on both instances (located on another server). Even though I do a "Scan files", the thumbnail does not show in the "other" instance... Here is a picture example...
[image]

I don't know if this is fixable. Maybe if the "thumbnail-storage-directory" could be set manually in a configuration option?
The first instance uses : ~/.octoprint/data/prusaslicerthumbnails/
and the second uses: ~/.octoprint2/data/prusaslicerthumbnails/
And that's ok, but the thumbs only get's "created" on upload... ?

jneilliii, 2020-05-05:
Yeah, not sure if I can fix this unless the data folder is shared between the two.

jneilliii, 2020-05-05:
Just pinged @foosel and she says this is not a good idea. 

> that's definitely a bad idea uploads folder should never be shared, messes up housekeeping badly

closed 2020-05-05 by jneilliii