jneilliii/OctoPrint-PrusaSlicerThumbnails #34: Newly uploaded gcode files missing show thumbnail button

steve220, 2020-05-07:
After installing the latest version, the show thumbnail button doesn't appear after uploading a new gcode file. If one hits the scan files button, the button subsequently appears.

jneilliii, 2020-05-07:
That's unexpected. Are there any errors in your octoprint.log or the developer console in your browser?

steve220, 2020-05-07:
I'll check this evening. Here is a screen shot. The preview shows in the large window, but the button is missing. Once I've looked for the browser errors, I'll update this thread.
[image]

jneilliii, 2020-05-08:
You mean this button?

[image]

jneilliii, 2020-05-08:
Oh I see, it's not showing on the file that's printing. Let me check.

jneilliii, 2020-05-08:
Yeah, so I just tested with the latest version freshly installed and I can't reproduce this issue. I even tried with the option to auto start the print and it's still showing up for me.

jneilliii, 2020-05-08:
This would be a stab in the dark, but maybe a file size issue?

steve220, 2020-05-08:
My browser isn't showing any errors in the console. As for the file size, you might be on to something! Smaller sized gcodes show up immediately. I did a quick experiment using a model that has two parts. On part's gcode is 55 meg, the other is 12 meg. Loading the 12 meg file shows the button right away, loading the large one doesn't show the button. Funny thing is, if I load the large one (button doesn't show), then I load the small one, the button shows on both files.  I've attached a zip of the gcode files I used for the test.
[catwoman.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4598920/catwoman.zip)

jneilliii, 2020-05-08:
Yeah, this is weird. I just tested on my development windows machine, and my pi with the larger file and the button is showing right after the saving process completes. I'm curious if you try the upload and the button doesn't show, if you refresh the file list does it then show?

Lanovar, 2020-05-22:
Im having the same issue. After updating to 0.1.1 - newly sliced gcodes dont show a thumbnail. 
The gcode file still contains "; thumbnail begin 16x16 832". 

Now to the strange part: 
Im running multiple instances of OP on one Pi. On the first printer (Prusa) it doesnt show the thumbnail, but on the second printer (Anycubic) it does. I sliced them with different printer profiles (both have thumbs enabled).

If i upload the "Anycubic gcode" to the Prusa - the thumb is showing.  
Maybe this helps finding the issue. 

[gcode.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4667911/gcode.zip)

jneilliii, 2020-05-22:
@steve220 can you let me know if the last version update resolved your issue? It should have resolved the problem with the extremely large file being processed.

steve220, 2020-05-22:
Yes! Thank you!

Steve

closed 2020-05-23 by jneilliii