jneilliii/OctoPrint-PrusaSlicerThumbnails #35: no button no change 0.4 to 1.0

BioCore01, 2020-05-08:
I've been trying since vesion 0.4 to 1.0 never had any change in the interface or button.

I've reinstalled and erased all data, including my print files. 
[image]
[image]
[image]

BioCore01, 2020-05-08:
my actif plugings are :

Action Command Prompt Support  
Action Commands (0.2)
Announcement Plugin  
Application Keys Plugin  
Backup & Restore  
Core Wizard  
Discovery  
Firmware Updater (1.7.0)
Fullscreen Plugin (0.0.5)
Logging  
Navbar Temperature Plugin (0.14)
OctoPod Plugin (0.2.6)
Pi Support Plugin  
Printer Safety Check  
Prusa ETA override Plugin (0.1.1) 
PrusaSlicer Thumbnails (0.1.0)
Software Update  
Themeify (1.2.2)
Virtual Printer

jneilliii, 2020-05-08:
Can you share one of your gcode files please?

BioCore01, 2020-05-08:
downloaded from my octoprint

https://www.swisstransfer.com/d/b4d4a0d1-81dc-49e4-a23f-370fbd340d75

BioCore01, 2020-05-08:
downloaded from my octoprint

[separation pour graine V1.0_0.3mm_ABS_MK3S_19m.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4599170/separation.pour.graine.V1.0_0.3mm_ABS_MK3S_19m.zip)

jneilliii, 2020-05-08:
Your gcode doesn't include the thumbnails inside the file, so the plugin can't extract them. Did you update your printer's ini file to include them and restart PrusaSlicer?

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails#configuration

BioCore01, 2020-05-08:
Okay, I forgot the most important step.

[image]

-->

[image]

--> add the part selected in blue in the .ini file

[image]

--> reload your slicer and upload your new code

[image]

BioCore01, 2020-05-08:
beautiful, now all working.

[image]

closed 2020-05-08 by BioCore01