jneilliii/OctoPrint-PrusaSlicerThumbnails #37: No thumbnail rendered, memory error in octoprint.log

ochm, 2020-05-09:
I uploaded big (128MB) .gcode file, and no picture is shown in Octoprint.
In octoprint.log i found:
_2020-05-09 05:28:09,509 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/skull_0.6n_0.15mm_PETG.gcode
2020-05-09 05:28:12,068 - octoprint.plugin - ERROR - Error while calling plugin prusaslicerthumbnails
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 80, in on_event
    self._extract_thumbnail(gcode_filename, thumbnail_filename)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 59, in _extract_thumbnail
    test_str = test_str.replace(octoprint.util.to_native_str('\r\n'),octoprint.util.to_native_str('\n'))
**MemoryError**_

Current free:

_# free
              total        used        free      shared  buff/cache   available
Mem:         997372      225356      576128        8392      195888      703684
Swap:        102396       67072       35324_

Zipped whole gcode is too big to be attached (34MB).
I made only thumbnail fragment
[thumbnail-only-skull_0.6n_0.15mm_PETG.gcode.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4602587/thumbnail-only-skull_0.6n_0.15mm_PETG.gcode.zip)

erezge, 2020-05-09:
This has to do with a change I requested in order to support also thumbnails in Cura which end with CRLF instead of LF like PrusaSlicer does.
After looking for a possible solution to remove the need for the replace I came up with this solution:
Replace : 
regex = r"(?:^; thumbnail begin \d+x\d+ \d+)(?:\n|\r\n?)((?:.+(?:\n|\r\n?))+)(?:^; thumbnail end)"  
With : 
regex = r"(?:^; thumbnail begin \d+x\d+ \d+)(?:\n|\r\n?)((?:[^\n\r]+(?:\n|\r\n?))+)(?:^; thumbnail end)"

This allows supporting both LF and CRLF without the replace statement. From my tests the solution seems to produce the exact same outputs.

jneilliii, 2020-05-09:
What version of octoprint?

jneilliii, 2020-05-09:
I probably need to come up with another parsing routine since it currently uses a regular expression match against the whole file contents. That's really overkill since the thumbnails are currently stored right at the very beginning of the file.

ochm, 2020-05-10:
> 
> 
> What version of octoprint?

OctoPrint 1.4.0 running on OctoPi 0.17.0

jneilliii, 2020-05-16:
I think I may have just fixed this issue with the latest release. Please check it out and let me know, version 0..1.1. You may have to force a check in the bottom of Software Update in the Advanced section at the bottom..

ochm, 2020-05-17:
Perfect! 0.1.1 renders the big gcode file without issue. Thank you!

closed 2020-05-17 by ochm