jneilliii/OctoPrint-PrusaSlicerThumbnails #38: No button available - maybe due to Canvas Plugin

bash-t, 2020-05-13:
I'm running OctoPrint 1.4.0 (@ OctoPi 0.16.0).

Installing PrusaSlicer Thumbnails 0.1.0 doesn't show up the button for displaying the thumbnail:

[image]

As additional info: I'm running the Canvas Plugin 2.3.3 since I have the Palette2. This plugin also modified the UI, by adding additional information to the file list.

Best regards,
Bastian

closed 2020-05-13 by bash-t

jneilliii, 2020-05-13:
I guess you figured out that you have to modify your printer profile in PrusaSlicer to allow the thumbnail to be included.

bash-t, 2020-05-13:
Hi jneilliii, you are right. That I figured out immediately after posting this issue.
Forgot to read everything to the bottom.  
Thanks, now it works like a charm! :)