jneilliii/OctoPrint-PrusaSlicerThumbnails #40: Multiple instances don't show thumbnail

jneilliii, 2020-05-22:
Im having the same issue. After updating to 0.1.1 - newly sliced gcodes dont show a thumbnail. 
The gcode file still contains "; thumbnail begin 16x16 832". 

Now to the strange part: 
Im running multiple instances of OP on one Pi. On the first printer (Prusa) it doesnt show the thumbnail, but on the second printer (Anycubic) it does. I sliced them with different printer profiles (both have thumbs enabled).

If i upload the "Anycubic gcode" to the Prusa - the thumb is showing.  
Maybe this helps finding the issue. 

[gcode.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4667911/gcode.zip)

_Originally posted by @Lanovar in https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/issues/34#issuecomment-632660666_

jneilliii, 2020-05-22:
@Lanovar, I've split this away as it's a separate issue. Do you share the same base folder on your multi instance setup?

Lanovar, 2020-05-22:
@jneilliii Sorry to ask, but where can i check this? Im not very good at this linuxstuff and followed a tutorial.

jneilliii, 2020-05-22:
The easiest way to know is if you upload a gcode file to one instance, does the file also show up in the other?

Lanovar, 2020-05-22:
Ah ok, makes sense. No it doesnt. But plugins are shared.

jneilliii, 2020-05-22:
Now I'm confused. Can you share the link to the video you followed for setting this up so I can test locally?

Lanovar, 2020-05-22:
Sure, here you go: https://www.youtube.com/watch?v=7Saa1HpLRoM

jneilliii, 2020-05-22:
This is weird. For some reason your prusa file is not able to be processed. It's giving the following error for me in octoprint.log. Check in the logging section of OctoPrint's settings and see if you see the same thing.

```
2020-05-22 21:53:43,660 - octoprint.plugin - ERROR - Error while calling plugin prusaslicerthumbnails
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 92, in on_event
    self._extract_thumbnail(gcode_filename, thumbnail_filename)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 67, in _extract_thumbnail
    if line.startswith(";") or line.startswith("\n"):
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 36: ordinal not in range(128)
```

jneilliii, 2020-05-22:
Try this version @Lanovar, just install using the browse button in Plugin Manager and then after restarting octoprint and reloading the page go into the plugin's settings and run the file scan.

[OctoPrint-PrusaSlicerThumbnails.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4670467/OctoPrint-PrusaSlicerThumbnails.zip)

Lanovar, 2020-05-23:
@jneilliii I sliced some stl's and it seems like its working again. Thx :)

closed 2020-05-24 by jneilliii