jneilliii/OctoPrint-PrusaSlicerThumbnails #41: exception parsing thumbnails

blalor, 2020-05-22:
I stopped seeing thumbnails at some point, and I see this in OctoPrint's log on a freshly-uploaded file:

```
2020-05-22 16:21:31,130 - octoprint.filemanager.analysis - INFO - Starting analysis of local:2020-05-22_NinjaFlex_3D_Benchy_0.2mm_FLEX_MK3S_2h27m.gcode
2020-05-22 16:21:31,131 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /opt/octoprint-venv/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /run/octoprint/uploads/2020-05-22_NinjaFlex_3D_Benchy_0.2mm_FLEX_MK3S_2h27m.gcode
2020-05-22 16:21:31,175 - octoprint.plugin - ERROR - Error while calling plugin prusaslicerthumbnails
Traceback (most recent call last):
  File "/opt/octoprint-venv/lib/python3.8/site-packages/OctoPrint-1.4.0-py3.8.egg/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/opt/octoprint-venv/lib/python3.8/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 92, in on_event
    self._extract_thumbnail(gcode_filename, thumbnail_filename)
  File "/opt/octoprint-venv/lib/python3.8/site-packages/octoprint_prusaslicerthumbnails/__init__.py", line 62, in _extract_thumbnail
    gcode = octoprint.util.comm.gcode_command_for_cmd(line)
  File "/opt/octoprint-venv/lib/python3.8/site-packages/OctoPrint-1.4.0-py3.8.egg/octoprint/util/comm.py", line 4739, in gcode_command_for_cmd
    gcode, _ = gcode_and_subcode_for_cmd(cmd)
  File "/opt/octoprint-venv/lib/python3.8/site-packages/OctoPrint-1.4.0-py3.8.egg/octoprint/util/comm.py", line 4747, in gcode_and_subcode_for_cmd
    match = regex_command.search(cmd)
TypeError: cannot use a string pattern on a bytes-like object
```

[2020-05-22_NinjaFlex_3D_Benchy_0.2mm_FLEX_MK3S_2h27m.gcode.gz](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4669317/2020-05-22_NinjaFlex_3D_Benchy_0.2mm_FLEX_MK3S_2h27m.gcode.gz)

This may be related to #40 and #34?

jneilliii, 2020-05-22:
Have you updated to python 3?

jneilliii, 2020-05-22:
Never mind, I see you are. I'll have to dig into this one.

jneilliii, 2020-05-22:
@blalor, I think the fix for #40 may actually resolve the issue for you in Python 3 as well. If you don't mind trying this version attached and let me know that would be greatly appreciated. Just use the browse button in plugin manager and select the downloaded zip file and restart octoprint/reload page.

[OctoPrint-PrusaSlicerThumbnails.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/4670549/OctoPrint-PrusaSlicerThumbnails.zip)

blalor, 2020-05-22:
This looks good!  I uploaded a new file from PrusaSlicer to OctoPrint and the thumbnail is visible.  Older gcode files with thumbnails weren’t processed, however, if that’s relevant.

jneilliii, 2020-05-23:
If you go into the plugin's settings and press the scan files button is it able to get those previously uploaded?

blalor, 2020-05-23:
Yep, looks like it is!

closed 2020-05-24 by jneilliii