jneilliii/OctoPrint-PrusaSlicerThumbnails #42: Most recent update removed thumbnails

KramRevilo, 2020-06-02:
After the most recent update my thumbnails are intermittently working.

KramRevilo, 2020-06-02:
Please let me know what information I can provide to help debug this.

jneilliii, 2020-06-02:
Please provide example gcode files that aren't working. Have you attempted to use the scan files button is the plugin's settings? It would be odd that this would be working intermittently unless some other factor is coming into play. Any other changes besides upgrading to the latest version?.

KramRevilo, 2020-06-02:
Okay, for some reason the thumbnails did not automatically populate BUT when I re-scanned through the 'scan files' button then it picked them up.  My apologies, I'll close this.

closed 2020-06-02 by KramRevilo