jneilliii/OctoPrint-PrusaSlicerThumbnails #43: State Thumbnail Not Updated

pearson, 2020-06-02:
The thumbnail displayed in the State panel does not update if a gcode file of the same name is uploaded. The new gcode will be printed if the State panel's print button is pressed, but the State panel thumbnail will remain out of date.

Steps to reproduce:
1) Upload a gcode file
2) Start a print
3) Cancel the print
4) Upload a different gcode file under the same name
5) Notice the difference between the thumbnails in the State panel and the Files panel.

An example is shown here, where a model was cut, the new STL sliced, and the resulting gcode uploaded under the same name:
[image]

jneilliii, 2020-06-02:
Thanks, this is a similar issue to deleting a gcode that is selected, the thumbnail does not get removed and stays loaded in view. I'll see about fixing this somehow in a future update.

ManuelMcLure, 2020-11-09:
I've noticed that if I select a file from the list the thumbnail appears in the State area, but if I then refresh the page the thumbnail in the State area goes away until I select a new file.
In case it matters, I'm using Themeify with the Discorded theme, as well as the OctoPrint-WideScreen plugin with the State and safety check moved to the right.

jneilliii, 2020-11-10:
I just tested this @ManuelMcLure and am not able to reproduce the refresh issue without widescreen plugin, so this might be a conflict somehow there. Might be too many UI hacks at the same time to the same elements maybe. I have Consolidated Tabs and Themeify so pretty sure we can rule out the theme being a problem. Other than that it works? I wonder if I fixed something in the 1.3.1rc1 that I'm already running that did something. You can give that version a try and report back that would be great. Use the URL below in plugin manager > Get More > ...from URL.

```
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/0.1.3rc1.zip
```

ManuelMcLure, 2020-11-10:
Just tested with 0.1.3rc1 and am still seeing the same problem. Also, disabling Octoprint-WideScreen doesn't seem to fix it either. I then went through my plugin list and disabled every plugin except for PrusaSlicer Thumbnails and the bundled plugins and I still see the issue.

jneilliii, 2020-11-10:
That's strange. are there any errors in the developer console of your browser?

jneilliii, 2020-11-10:
Let me make sure I have the steps right. Select the file, it loads the thumbnail in the state panel. Refresh the page...thumbnail is gone. It's not the same issue if you're printing?

ManuelMcLure, 2020-11-10:
Even if I'm printing. I first noticed it when I was connecting with a different browser - even though the file was loaded and printing the new browser window would not show the thumbnail in the state panel. Then I figured out that I could reproduce the problem with a single browser by refreshing. I'll get you the info from the developer console in a sec, but I am seeing this on Chrome and Edge on Windows 10 as well as Safari on iPhone, so it doesn't seem browser-specific.

ManuelMcLure, 2020-11-10:
The only error in the developer console after a refresh is:
```
packed_core.js:872 Could not bind view model ETAModel to target #settings_plugin_display_eta : Error: You cannot apply bindings multiple times to the same element.
    at m (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:8070)
    at k (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7607)
    at g (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://oprint.internal.mclure.org/static/webassets/packed_libs.js?df208d52:3:7660)
```

ManuelMcLure, 2020-11-10:
Inspecting the state panel shows that the `<div id="prusaslicer_state_thumbnail"...` is completely gone after the refresh.

jneilliii, 2020-11-10:
strange. 

[image]

ManuelMcLure, 2020-11-10:
I'm not sure what you're seeing - it looks like collapsing the Connection is causing the thumbnail to reappear?

ManuelMcLure, 2020-11-10:
[image]

jneilliii, 2020-11-10:
The connection panel closes on it's own on page load over here. So not sure why that would cause any difference or why yours appears to be closed for the page even starts to render. Maybe it's my power plugins causing enough of a delay on load. Let me do some more testing on my end. If I can get it to a point of being able to reproduce the issue I'll see what I can do to fix it. You said it happens even with all other plugins turned off? Just trying to think of the other plugins that might be modifying the state panel. Looks like you have DLP?

ManuelMcLure, 2020-11-10:
I have DLP installed but disabled. Other than that I have:
- Active Filters Extended
- Bed Visualizer
- Detailed Progress
- Display Z
- Firmware Updater
- Fullscreen Plugin
- M73 Progress Plugin
- Navbar Temperature Plugin
- OctoPod Plugin
- OctoPrint-Display-ETA
- OctoPrint-Widescreen
- OctoPrint-WiFiStatus
- PrintJobHistory
- PrintTimeGenius Plugin
- PrusaSlicer Thumbnails
- PSU Control
- Themeify
- TouchUI (disabled)

I just tested again disabling all of them except for PrusaSlicer Thumbnails and see the same problem.

ManuelMcLure, 2020-11-10:
A complete fresh install of OctoPrint with basic settings and just the thumbnail plugin added works like your video, so it's something to do with my settings, apparently.

ManuelMcLure, 2020-11-10:
Definitely a settings thing. If I replace my backed up `.octoprint` folder it starts failing again. I'll do some more digging to see if I can see what's triggering this.

ManuelMcLure, 2020-11-10:
Apparently this has something to do with `groups.yaml`. If I delete that file from `.octoprint` the problem goes away. So this is something specific to my configuration.
After deleting `groups.yaml` I then added "Webcam" and "Status" permissions to the "Guests" group and the problem started happening again. Deleting those two permissions fixed it again.

jneilliii, 2020-11-10:
What if you add the File List permission along with status?

jneilliii, 2020-11-10:
nope, that doesn't work, but I am able to reproduce your issue now with enabling status to the guests group. I'll try to figure out why that happens.

ManuelMcLure, 2020-11-10:
Same problem with File List + Status. Note that I'm logged in as a user that's a member of Admins and Operator so the permissions of Guest shouldn't really apply.

jneilliii, 2020-11-10:
yeah, I'm seeing the same thing. I do not know why changing those options is effecting the display of the thumbnail at all, like you said logged in as admin shouldn't be effected. I'll try to figure it out.

jneilliii, 2021-04-17:
This is resolved with the above commit and available in version 0.1.5rc7.

closed 2021-04-17 by jneilliii