jneilliii/OctoPrint-PrusaSlicerThumbnails #44: Thumbnail gone on octoprint instances with a shared upload folder

Connor2811, 2020-06-03:
I have 5 printers running off one server with 5 octoprint instances. They share an upload folder but the g-code is only sent to one instance's API key. 

when I upload, the printer with the API key loads the thumbnail no issues but all other instances show a small place holder photo.
[image]
[image]

jneilliii, 2020-06-03:
Interesting use case, can't say I've done much with multiple instances. The file gets uploaded to the plugin's data folder of the working instance. If that data folder is not also shared between instances then this could happen. You could manually sync the folders outside of octoprint potentially to get around this or use a symlink(?) to the plugin's subfolder inside the `.octoprint/data` folder of the instance you upload to.

Connor2811, 2020-06-03:
just tried this, it doesnt look like it syncs properly, the files show up but no thumbnails are displayed

jneilliii, 2020-06-03:
Is the plugin installed on all instances? Can you please go to one of the broken instances and then append `/api/files/local` to the end of the URL so it looks like the example below and post the returned json?

http://192.168.0.2:5001/api/files/local

Connor2811, 2020-06-03:
`{"files":[{"date":1591136852,"display":"Extruder Motor Mount VlostCount _0.2mm_PLA_MK2.5S_10h57m.gcode","gcodeAnalysis":{"dimensions":{"depth":141.98,"height":82.0,"width":201.239},"estimatedPrintTime":31224.709187547152,"filament":{"tool0":{"length":41746.2325300373,"volume":100.41145647092819}},"printingArea":{"maxX":201.239,"maxY":138.98,"maxZ":82.0,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"a849f9cdc20c2b0ccbd5d1a75bf249abcb39cfa4","name":"Extruder_Motor_Mount_VlostCount__0.2mm_PLA_MK2.5S_10h57m.gcode","origin":"local","path":"Extruder_Motor_Mount_VlostCount__0.2mm_PLA_MK2.5S_10h57m.gcode","refs":{"download":"http://10.0.0.209:5002/downloads/files/local/Extruder_Motor_Mount_VlostCount__0.2mm_PLA_MK2.5S_10h57m.gcode","resource":"http://10.0.0.209:5002/api/files/local/Extruder_Motor_Mount_VlostCount__0.2mm_PLA_MK2.5S_10h57m.gcode"},"size":20864819,"type":"machinecode","typePath":["machinecode","gcode"]},{"date":1591177443,"display":"test trim.gcode","gcodeAnalysis":{"dimensions":{"depth":157.241,"height":83.4,"width":191.846},"estimatedPrintTime":49132.162891641,"filament":{"tool0":{"length":77160.3666397687,"volume":185.592431378216}},"printingArea":{"maxX":191.846,"maxY":154.241,"maxZ":83.4,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"2afb3a257b98e9e997b3c3eb202e64ffe8cf5fbf","name":"test_trim.gcode","origin":"local","path":"test_trim.gcode","refs":{"download":"http://10.0.0.209:5002/downloads/files/local/test_trim.gcode","resource":"http://10.0.0.209:5002/api/files/local/test_trim.gcode"},"size":51124408,"thumbnail":"plugin/prusaslicerthumbnails/thumbnail/test_trim.png?20200603024403","thumbnail_src":"prusaslicerthumbnails","type":"machinecode","typePath":["machinecode","gcode"]},{"date":1591045978,"display":"BaselineDesignMask V4- Clamp Exterior V2_0.2mm_PLA_MK2.5_1h59m.gcode","gcodeAnalysis":{"dimensions":{"depth":181.42,"height":8.6,"width":195.309},"estimatedPrintTime":5154.52852647178,"filament":{"tool0":{"length":10374.408919999536,"volume":24.95337774331158}},"printingArea":{"maxX":195.309,"maxY":178.42,"maxZ":8.6,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"e1b70d1ee8a0223582954a8fbe39f35f0a9ea36f","name":"BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.gcode","origin":"local","path":"BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.gcode","prints":{"failure":0,"last":{"date":1591053282.536474,"printTime":7303.707501418998,"success":true},"success":1},"refs":{"download":"http://10.0.0.209:5002/downloads/files/local/BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.gcode","resource":"http://10.0.0.209:5002/api/files/local/BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.gcode"},"size":4917117,"statistics":{"averagePrintTime":{"_default":7303.707501418998},"lastPrintTime":{"_default":7303.707501418998}},"thumbnail":"plugin/prusaslicerthumbnails/thumbnail/BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.png?20200601141258","thumbnail_src":"prusaslicerthumbnails","type":"machinecode","typePath":["machinecode","gcode"]},{"date":1591180181,"display":"test69.gcode","gcodeAnalysis":{"dimensions":{"depth":157.241,"height":83.4,"width":191.846},"estimatedPrintTime":49132.162891641,"filament":{"tool0":{"length":77160.3666397687,"volume":185.592431378216}},"printingArea":{"maxX":191.846,"maxY":154.241,"maxZ":83.4,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"2afb3a257b98e9e997b3c3eb202e64ffe8cf5fbf","name":"test69.gcode","origin":"local","path":"test69.gcode","refs":{"download":"http://10.0.0.209:5002/downloads/files/local/test69.gcode","resource":"http://10.0.0.209:5002/api/files/local/test69.gcode"},"size":51124408,"thumbnail":"plugin/prusaslicerthumbnails/thumbnail/test69.png?20200603032941","thumbnail_src":"prusaslicerthumbnails","type":"machinecode","typePath":["machinecode","gcode"]},{"date":1591176332,"display":"Trim Corner_0.2mm_PLA_MK2.5S_17h21m.gcode","gcodeAnalysis":{"dimensions":{"depth":157.241,"height":83.4,"width":191.846},"estimatedPrintTime":49132.162891641,"filament":{"tool0":{"length":77160.3666397687,"volume":185.592431378216}},"printingArea":{"maxX":191.846,"maxY":154.241,"maxZ":83.4,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"2afb3a257b98e9e997b3c3eb202e64ffe8cf5fbf","name":"Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.gcode","origin":"local","path":"Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.gcode","refs":{"download":"http://10.0.0.209:5002/downloads/files/local/Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.gcode","resource":"http://10.0.0.209:5002/api/files/local/Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.gcode"},"size":51124408,"thumbnail":"plugin/prusaslicerthumbnails/thumbnail/Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.png?20200603022532","thumbnail_src":"prusaslicerthumbnails","type":"machinecode","typePath":["machinecode","gcode"]},{"date":1591009316,"display":"BaselineDesign Mask-  toggle .6in Toggle test_0.2mm_PLA_MK2.5_21m.gcode","gcodeAnalysis":{"dimensions":{"depth":143.204,"height":8.6,"width":140.705},"estimatedPrintTime":921.1390004640027,"filament":{"tool0":{"length":919.821439999979,"volume":2.212429838240588}},"printingArea":{"maxX":140.705,"maxY":140.204,"maxZ":8.6,"minX":0.0,"minY":-3.0,"minZ":0.0}},"hash":"2487eefebbbd8aa852e1dae23baeb7cb95d1c918","name":"BaselineDesign_Mask-__toggle_.6in_Toggle_test_0.2mm_PLA_MK2.5_21m.gcode","origin":"local","path":"BaselineDesign_Mask-__toggle_.6in_Toggle_test_0.2mm_PLA_MK2.5_21m.gcode","prints":{"failure":0,"last":{"date":1591012976.012288,"printTime":1470.067613407,"success":true},"succes [... cut ...] MK2.5_30m.gcode","refs":{"download":"http://10.0.0.209:5002/downloads/files/local/Plunger_Mount_V2_Part3_0.2mm_PLA_MK2.5_30m.gcode","resource":"http://10.0.0.209:5002/api/files/local/Plunger_Mount_V2_Part3_0.2mm_PLA_MK2.5_30m.gcode"},"size":393651,"thumbnail":"plugin/prusaslicerthumbnails/thumbnail/Plunger_Mount_V2_Part3_0.2mm_PLA_MK2.5_30m.png?20200602153604","thumbnail_src":"prusaslicerthumbnails","type":"machinecode","typePath":["machinecode","gcode"]}],"free":6501707776,"total":12610314240}`

jneilliii, 2020-06-04:
Thank you so much, so based on that the following gcode were uploaded with thumbnails.

plugin/prusaslicerthumbnails/thumbnail/test_trim.png
plugin/prusaslicerthumbnails/thumbnail/BaselineDesignMask_V4-_Clamp_Exterior_V2_0.2mm_PLA_MK2.5_1h59m.png
plugin/prusaslicerthumbnails/thumbnail/test69.png
plugin/prusaslicerthumbnails/thumbnail/Trim_Corner_0.2mm_PLA_MK2.5S_17h21m.png
plugin/prusaslicerthumbnails/thumbnail/BaselineDesign_Mask-__toggle_.6in_Toggle_test_0.2mm_PLA_MK2.5_21m.png
plugin/prusaslicerthumbnails/thumbnail/Plunger_Mount_V2_Part1_0.2mm_PLA_MK2.5_5h53m.png
plugin/prusaslicerthumbnails/thumbnail/Plunger_Mount_V2_Part3_0.2mm_PLA_MK2.5_30m.png

Does that look about right? If so can you tell me if those images are in the `~/.octoprint/data/prusaslicerthumbnails/` folder for both instances? If you go to the address http://10.0.0.209:5001/plugin/prusaslicerthumbnails/thumbnail/test_trim.png does it not load the image? I'm curious if the broken image might be a caching problem.

Connor2811, 2020-06-04:
Yep file loads no issue at that url. Do you think it's a caching issue on the client side or the server side?

jneilliii, 2020-06-04:
I would think client side, that was one of the reasons I added the timestamp to the end of those URLs in the metadata for when files get re-uploaded. Do a forced refresh using shift+f5 and see if they load.

Connor2811, 2020-06-05:
tried that, still doesnt load for the other instances. what I find really weird is that some of the earlier files have loaded but I'm not sure when.

jneilliii, 2020-06-05:
So weird, what happens if you go into settings on the broken instance and click scan files? Does it break the currently working instance?

Connor2811, 2020-06-06:
oh, that's actually super interesting. when I click scan it gives me an error that says "no_thumbnail" but when I close settings it loads the thumbnail no issue

Connor2811, 2020-06-06:
[image]

jneilliii, 2020-06-06:
Ok, it's probably the metadata refresh more than likely. Try this on the next upload to one instance. After it's finished and is visible in the instance uploaded to, let the file sync/copy the thumbnail from data/prusaslicerthumbnail folder to the other instance (whatever process you setup before) and then in the other instance press the refresh files list button in the file list.

Connor2811, 2020-06-07:
Im using a symlink in rasbian, the folder I upload to is the main folder and then all other instances use a symlinked folder in their .octoprint folder. maybe a setting to change which folder your plugin looks at? sorry I'm really new to Linux and server systems.

if I upload a file and then wait a couple of hours then refresh files list nothing happens

Connor2811, 2020-06-07:
What exactly happens when you press scan files? Is there a way to run the same thing when the "refresh files list" button is pressed?

jneilliii, 2020-06-07:
So when you press the scan files button it's basically grabbing a list of metadata provided by OctoPrint and then checking to see if the thumbnail attribute of the file exists. If it doesn't it tries to extract the thumbnail to the plugin's data folder and then sets the attribute. If the attribute is already there it doesn't do anything. Once all the files have bee processed it's forcing a file list update, which is the same thing as pressing that button in the file list. 

If you can provide the details of how you setup the multiple instance, with specifics of your symlinks I can reproduce on one of my development pis and see if I can figure out what's happening.

closed 2020-06-28 by stale[bot]