jneilliii/OctoPrint-PrusaSlicerThumbnails #45: wrong link in octoprint

Avatarsia, 2020-06-13:
when i click on the hompage button where i can download your plugin it leads to this page

 https://github.com/scottrini/OctoPrint-PrusaLevelingGuide

jneilliii, 2020-06-14:
Thanks for the heads up. That is not expected. Not sure why that is happening. Looking into it.

jneilliii, 2020-06-14:
Not sure where you are seeing this, all associated links seem correct from here. I searched each of the repositories and don't see the cross over anywhere.

https://github.com/OctoPrint/plugins.octoprint.org/search?q=PrusaLevelingGuide&unscoped_q=PrusaLevelingGuide

https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/search?q=PrusaSlicerThumbnails&unscoped_q=PrusaSlicerThumbnails

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/search?q=PrusaLevelingGuide&type=Code

Avatarsia, 2020-06-14:
hmmm i don't know what was wrong, i can't repeat the way without geting to the correct side.
so seems maybe  mistake of mine

closed 2020-06-14 by Avatarsia