jneilliii/OctoPrint-PrusaSlicerThumbnails #46: Access Thumbnail via API?

qJake, 2020-07-10:
First, I'm not even sure if, as part of the OctoPrint plugin system, you can even expose something via the REST API to begin with... that being said.

It would be awesome if there were a REST endpoint where I could fetch the thumbnail image of the current print.

Or maybe offer it up as an endpoint with a secure token on the end, like:

`http://192.168.0.2/api/plugins/thumbnails/currentjob.jpg?token=abcd1234abcd1234....`

I was hoping that it would be available in the `GET /api/job` call but no dice there.

Is this possible, and if so, could this be added in a future update?

jneilliii, 2020-07-10:
Several outside people ate using this, like octopod, printoid, and OctoFarm. It's a property in the file api return results.

qJake, 2020-07-11:
Oh! I was looking in `/api/job` and the `file` object returned there, not the `/api/files` endpoint. I see it now. 🙂 

For anyone else, calling `/api/files/local/Path/To/Your/file.gcode` returns a `thumbnail` property with a path to the image file.

Thanks @jneilliii!

closed 2020-07-11 by qJake