jneilliii/OctoPrint-PrusaSlicerThumbnails #49: [Feature Request] Add Print button and details to window when clicking on thumbnail in File panel

SigmaRelief, 2020-10-07:
First thanks for the plugin, it is a huge help keeping track of items in the queue.

I have inline thumbnails enabled and when I mouse over the image, the entire file "cell" including details and action buttons highlights implying some sort of global action, but a click only brings up an enlarged version of the image itself.  It would be extremely convenient if this window included action buttons (posibly a large print button) and all of the extended file details as  part of a purpose made screen.  A small extension would be arrow buttons (and or Up/down keyboard shortcuts) to scroll through all files while in this popout window with the larger image size to clearly see differences in files.

With inline thumbnails displayed, only one file fits in the Files pane at a time and then only with careful alignment and the extended details hidden, breaking everything out to a larger window makes reading this easier.   An option (probably a separate feature request) to extend the height of the Files panel would also be a useful adaptaiton to better show off the thumbnails.

jneilliii, 2020-11-07:
* [ ] file details and print button in pop-up
* [ ] height adjustment for file list

For the later I've used Themeify since I use that and was able to easily adjust the height of the file list. I will look into incorporating the above two items to the plugin's settings. One thing that might assist you is scaling the image to 50% and centering it in the inline view in the plugin's settings to show more in the interim. Cycling through the file list is something that I believe to be too much effort for the reward just thinking of it in my head and how that pop-up works.

jneilliii, 2021-02-25:
I've added the print button and file list height options in the commit above. You can test using the Release Candidate channel in Software Update settings if you want to try it out.

closed 2021-02-28 by jneilliii