jneilliii/OctoPrint-PrusaSlicerThumbnails #50: Thumbnail State not showing, only thumbnail inline in files

chrislolds, 2020-10-14:
I'm using two octoprint instances, one for a Prusa Mini and one for an MK3S, the Mini works fine and displays the thumbnail in the State and in the Files inline, the MK3S only shows the thumbnail inline in the Files and not the State.

I've made the modification to the printer ini in Slic3r PE, have the newest version of Slicer, newest version of Octoprint with Python3, newest version of the Thumbnail Plugin and have rebooted the OctoPi and flagged and reflagged the setting, the gcode has the thumbnail in it and it is showing up in the file but I can not get it to show in the Printer State.

Any ideas?

jneilliii, 2020-10-14:
multiple instances of OctoPrint running on the same device? It should work, but I'll need more details as to your setup. If you can provide guides on how you configured things that would be helpful as well so I can re-create your environment.

chrislolds, 2020-10-14:
> multiple instances of OctoPrint running on the same device? It should work, but I'll need more details as to your setup. If you can provide guides on how you configured things that would be helpful as well so I can re-create your environment.

I apologize, 2 different rPi's, was just mentioning that it's working out of the box as expected with the Mini and that OctoPi but not with a different OctoPi and the MK3S.

Using the default MK3S profile and other than an MMU2 Filament Selector I am running the same plugins as well.  I've tried disabling plugins but can't figure out what would be blocking the Printer State thumbnail from showing.  Wasn't sure if this was an issue others had seen previously?

chrislolds, 2020-10-14:
I was just looking through what was different about the two installs and on my MK3S install I have your Wemo Switch plugin, I disabled that and it is now showing the thumbnail.  I am uninstalling, cleaning and will be reinstalling and configuring the Wemo Switch plugin and will let you know if the thumbnail works or if it has a compatibility issue with the Wemo Switch plugin.

Thank you for your help, it's amazing that a developer creates amazing plugins for the community to use for free and is also so responsive at bugs and helping people with those plugins, I really appreciate it!

chrislolds, 2020-10-14:
After reinstalling the Wemo Switch Plugin everything seems to be working correctly!  Please let me know if you'd like any other information from me, I'd be happy to help in anyway I can.

chrislolds, 2020-10-14:
Problem seems to have crept back in.  Currently an hour into an 11 hour print so I will test when I am done but the thumbnails were showing up for a few prints and have vanished again.  I will try disabling the Wemo Plugin and test that for awhile to see if I continue to have the issue.

I am using a Prusa MK3S with MMU2S and the print I am currently doing is Multi-material.  It is showing the thumbnail in the file view and I do have the MultiMaterial printer profile set with the thumbnail flag, this is the first MultiMaterial print I've done since resetting the Wemo Plugin earlier.

jneilliii, 2020-10-14:
This is odd, I can't think of any reason why there would be a conflict between these two plugins. Let me know if you think it's specific to the multi material gcode or the wemo plugin conflict after a little more testing. When it's not working try opening your developer console by pressing f12 on your keyboard and let me know if you see any errors.

jneilliii, 2020-11-07:
Any updates from your end on this @chrislolds as to why this might be happening?

closed 2021-02-09 by github-actions[bot]