jneilliii/OctoPrint-PrusaSlicerThumbnails #53: Greedy regex

Pavulon87, 2020-11-07:
For me current regex is too greedy.
When slicer generates small and big thumbnail in octoprint I see only the small one(first instead of last).
After a little of debugging I saw that len(matches) returns 1. Problem is that current regex match from start of first thumbnail to the end of second one.
Changing
`regex = r"(?:^; thumbnail begin \d+x\d+ \d+)(?:\n|\r\n?)((?:.+(?:\n|\r\n?))+)(?:^; thumbnail end)"`
to
`regex = r"(?:^; thumbnail begin \d+x\d+ \d+)(?:\n|\r\n?)((?:.+(?:\n|\r\n?))+?)(?:^; thumbnail end)"`
Fixes problem for me and shouldn't break anything.

jneilliii, 2020-11-07:
Thanks for this. I just noticed recently that SuperSlicer was introducing this for me as well and just haven't looked into it yet. I will give your change a go and see how well it works. You didn't by chance test with just one thumbnail did you?

Pavulon87, 2020-11-08:
No, I tested with two. I also use a superslicer, maybe it's his fault.

jneilliii, 2020-11-08:
I'm not 100% sure, but it could be related. Either way, I merged your changes into the 0.1.3rc1 version if you want to double-check it still works as expected you can install by copying/pasting the URL below in plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/0.1.3rc1.zip
```

Pavulon87, 2020-11-08:
It works for me. I don't see any problems.

closed 2020-11-08 by Pavulon87