jneilliii/OctoPrint-PrusaSlicerThumbnails #55: Thumbnails gone after moving from subfolder

sourekv, 2020-11-13:
I am using a DeleteAfterPrint Plugin. When I move file from archive folder back to root thumbnail is gone.
[image]

jneilliii, 2020-11-13:
That should get caught as I implemented work arounds for such a thing. If you go into the plugin's settings and press the scan files button it should get them back.

sourekv, 2020-11-14:
Scan Results
No convertible files found
:-(

jneilliii, 2020-11-14:
Ok, I'll look into it.

sourekv, 2020-12-05:
Hi, seems fixed in 0.1.3.

jneilliii, 2020-12-05:
Cool, closing for now. If you see this as an issue in the future, please just comment here and I'll re-open.

peasantdave, 2020-12-16:
I am still seeing this on OctoPrint 1.5.2 / PrusaSlicer Thumbnails 0.1.3. Was able to reproduce with: 

1. Have "object123.gcode" in root /  (Image URL: http://X.X.X.X/plugin/prusaslicerthumbnails/thumbnail/object123.png)
2. Move "object123.gcode" to /subfolder (Image URL: http://X.X.X.X/plugin/prusaslicerthumbnails/thumbnail/subfolder/object123.png)
3. Move "object123.gcode" back to root / and I see the broken image icon. If I right click and "Open image in new tab" I get "403: Forbidden". It seems the URL has an extra /. If I take out the extra / in the URL the image loads fine. (Image URL: http://X.X.X.X/plugin/prusaslicerthumbnails/thumbnail//object123.png)

jneilliii, 2020-12-16:
Thanks for tracking down the details. I have an idea why that might be happening moving back to root.

jneilliii, 2020-12-16:
I think I have this fixed with the above commit. I have created version 0.1.4rc1 to test. If you go into software update and change the drop-down to the Release Candidate channel you should get a notice to update to that version.

peasantdave, 2020-12-16:
Thanks. I just tested and it worked correctly now. 

For everything that was already broken in my root I had to move to another folder and move it back.

jneilliii, 2020-12-16:
Cool, thanks for double-checking. This will be included in the next release. Just need to see if there are any other changes that I can incorporate prior to official release.

sourekv, 2020-12-16:
Good job!

closed 2021-01-17 by jneilliii