jneilliii/OctoPrint-PrusaSlicerThumbnails #56: Octoprint 1.5.0 update and now PrusaSlicerThumbnails no longer show thumbnails in "State"

wdgeek, 2020-11-30:
Updated two separate Octoprint Pi machines, a MK3s and a Prusa Mini to Octoprint 1.5.0
After the update both machines no longer show thumbnails in State, but does show thumbnails in Files.
I disabled all my installed plugins for my Prusa Mini to see if any of them were interfering with your plug-in but still no thumbnails in the State window.

jneilliii, 2020-11-30:
Thanks, I've recently noticed that in my 1.5.0 release candidates as well and haven't had a chance to investigate further. I will be looking into it.

jneilliii, 2020-11-30:
Can you try with the RC version that i had queued up? Copy/paste the URL below into Plugin Manager > Get More > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/0.1.3rc2.zip
```

wdgeek, 2020-11-30:
I noticed the update rc just now. 
Updated it, and got a message that there was an "official" release.
Installed the official release.
Tried it and can also confirm that its working now on my MK3.
THANK YOU.

jneilliii, 2020-11-30:
Yeah, I rushed the release since this is going to effect a bunch of people. Thanks for confirming.

closed 2020-11-30 by jneilliii