jneilliii/OctoPrint-PrusaSlicerThumbnails #61: Actual Printing Thumbnail?

FlexerJR, 2021-01-19:
Hi jneilliii,

thanks for your great work on this plugin. I have a tablet on the wall which is showing my smarthome system. Yesterday i did some parsing of the thumbnail urls to get the link to the image of the actual print. That works great and i can now see the thumbnail of the print in my tablet. 

But there is a problem. I can not do a parsing of the thumbnails url in the subfolders. 

Is it possible to have one place for the actual printing thumbnail with is at a fixed place which will be updated when starting a new print even if the print file will be started out of a subfolder?

http://xxx.xxx.xxx.xxx/plugin/prusaslicerthumbnails/thumbnail/Actual/Cube.png?20201229224717

Would be very very nice!!! I show you a picture. 

[image]

Best regards

jneilliii, 2021-01-19:
The relative url to the thumbnail is actually contained within the files metadata. It is a property named thumbnail.

FlexerJR, 2021-01-20:
I just solved my issue in a other way. you can close this one. But Thanks for your response!

FlexerJR, 2021-01-20:
Closed by my side

closed 2021-01-20 by FlexerJR