jneilliii/OctoPrint-PrusaSlicerThumbnails #62: [issue/feature request] Thumbnails not showing on new files added

taibhse-designs, 2021-01-23:
Issue, 
when new gcode files are added / uploaded to octoprint, when the file viewer is refreshed, the gcode files show up but thumbnail is not present. If I go into plugin settings and scan for thumbnails, they then show up. Is it possible or could it be added for the plugin to auto generate thumbnails for new files added.

taibhse-designs, 2021-01-23:
To add to this, my setup is the latest octoprint installed on windows through virtualenv, paths to gcode folders are on the windows drive and set in octoprint

jneilliii, 2021-01-23:
The plugin is programmed to automatically add the thumbnail when uploaded through the octoprint interface or api (sent from Cura or PrusaSlicer\SuperSlicer). If it's not automatically then there could be a conflict with another plugin. Could you please share your octoprint.log so I can get a fuller view of your environment to have a better idea? You can just drag it into a comment here on github.

taibhse-designs, 2021-01-24:
OK just tested with the api and uploading that way, thumbnails auto generate now, though the original issue was if I manually save a gcode file over the network to the folder, using the refresh files button in octoprint doesnt auto generate a thumbnail, while the api and sending that way is useful, sometimes I have to save files out into dedicated subfolders and organise before printing, so its manually saving to the folder. It would be ideal if the plugin detected when a user refreshes the file list and uses that as a hint to search for new files to generate thumbnails also.

jneilliii, 2021-01-24:
That is because when you just save the file to the uploads folder of octoprint the standard file processing doesn't take place. Rather than saving into the uploads folder if you save the file to the watched folder it will get picked up by OctoPrint and all proper add file processing will take place.

bagobones, 2021-01-26:
Just updated to PrusaSlicer 2.3.0 and the build in upload function seems to have this issue. If I upload and print, I get no preview in status or in the files session. If I scan I get the update in the files section so I know the gcode is correct.

It has been a few months since I printed last and that was on Prusa Slicer 2.2.0. I am sure that it previewed on upload before.

jneilliii, 2021-01-26:
Are you using the option to auto start the print? Curious if that is possibly causing issues or some other plugin that is also conflicting/ processing files on upload? Do tou have any errors in octoprint.log?

bagobones, 2021-01-26:
Yes I am using auto start.. if I have time (currently have a print running) I will try after my next print.

I have the following 3rd Party plugins installed:

Pushover (0.3.8)
PrusaSlicer Thumbnails (0.1.4)
Prusa ETA override Plugin (0.1.1)
Filament Sensor Reloaded (1.3.4)
Cancel Objects (0.4.4)

Cancel Objects reads additional gcode so it could be a candidate.

jneilliii, 2021-01-26:
Yeah, I use auto start print too and haven't seen this happen for me. I don't have either of those 2 plugins installed, so there could be contention on file upload.

bagobones, 2021-01-26:
Just started an upload from PrusaSlicer without the auto start.. The preview appeared in the file list immediately, it also appeared in the Status when I selected it to print.

However I almost always use auto start and I am almost 100% sure this used to work however since the last time I remember it working I have updated PrusaSlicer, Octoprint and this plugin. So lots of variables.

jneilliii, 2021-01-27:
ok thanks. I haven't printed in a while and have definitely not attempted with the latest PrusaSlicer version as I use SuperSlicer personally. I'll try to do some testing this weekend to see if I can reproduce the issue. If you happen to see a recurring pattern other than the auto start option please let me know.

bagobones, 2021-01-28:
OK, odd ran another print today with the start option and it worked this time?

jneilliii, 2021-01-28:
That is very odd. It could potentially be race condition where it doesn't happen all the time, or possibly with certain size files?

bagobones, 2021-01-28:
If I can produce something repeatable I will share it.. Only keeping an eye for now. Just wanted to note it may be intermittent as I know how frustrating it is to reproduce those issues.

closed 2021-04-17 by jneilliii