jneilliii/OctoPrint-PrusaSlicerThumbnails #65: [FR] Scale large thumbnail in State section

lightmaster, 2021-02-25:
This plugin shows a large thumbnail at the top of the State section, which in my case causes the Print, Pause, and Cancel button to be pushed down a little bit so I have to scroll from the top of the page to see them. This plugin also already has a scaling feature for the inline thumbnails in the File List section. I propose to add a scaling option to the large thumbnail in the State section as well.

Also, would it be possible to relocate the State section thumbnail to the bottom of this section instead of the top, so important buttons will always be accessible when the page is scrolled to the top?

jneilliii, 2021-02-25:
I have no issue with adding the scaling option, or potentially just apply the same scaling value for inline thumbnails. I'll look at the other options as well as for where to place the thumbnail. That's something I'm looking into for the file list anyway.

lightmaster, 2021-02-25:
I'd say use 2 separate scaling options. Playing around with some custom CSS with UI Customizer until this gets implemented, and I'm liking 70% for the big thumbnail, whereas I use 25% for the file list so they are absolutely massive in the file list, lol.

jneilliii, 2021-02-25:
Available in 0.1.5rc2 release is you want to try it out.

lightmaster, 2021-02-25:
Is it possible to change the State Section thumbnail without having to reload the page? You can change the File Section thumbnails without having to reload, and my custom CSS workaround scaled the State Section thumbnail without a reload.

jneilliii, 2021-02-25:
Ah, yeah. I just need to add the knockout subscription on the size value.

jneilliii, 2021-02-25:
0.1.5rc3 out with subscription enabled.

lightmaster, 2021-02-25:
Its now scaling without a refresh. Thanks for the quick updates!

closed 2021-02-28 by jneilliii