jneilliii/OctoPrint-PrusaSlicerThumbnails #68: [Request] Move inline thumbnails to the additional info drop-down.

erispe, 2021-03-05:
This plugin is amazing! But the file browser panel is quite bad and space is precious. I wanted to suggest a change where the inline thumbnail is moved to the Additional Information dropdown arrow, so it can be hidden. I don't want to see all thumbnails all the time.
[image]

An alternative idea is to put the buttons layout into a new horizontal layout and put the thumbnail to the left, this uses all the available horizontal space instead of just stacking vertically.
[image]

jneilliii, 2021-03-05:
Yeah, I'm already working on this for my Cura Thumbnails plugin but haven't made much headway yet. I want to make it user configurable because not everyone uses Themeify (or other UI plugin) and has this wide of a sidebar.

erispe, 2021-03-05:
That's cool! I didn't know Cura has thumbnails too now. I just found this plugin so this was my immediate impression. Glad to see you are thinking about this too! In my opinion the bigger issue here is the super compact file browser. I would like to completely redesign it but I don't know anything about making plugins.

jneilliii, 2021-03-05:
I didn't either when I started, not even python and it kind of snowballed. 

https://plugins.octoprint.org/by_author/#jneilliii

There are other plugins like my Widescreen plugin, UI Customizer, or like in your case Themeify that end up making the sidebar wider so it's just a matter of figuring out a layout that works.

jneilliii, 2021-03-06:
So I have made a little progress on the left side method, that works well with UI Customizer (4 column sidebar left) and 40% scaling on the image. I'd also like to point out that I have a configurable file list height in my current RC version that makes the list a little easier to deal with, if you haven't already set up that in Themeify's custom settings.

[image]

jneilliii, 2021-03-06:
Added with the above commit. About to release 0.1.5rc3 that will have these changes in it. If you want to test it out, change the release channel to `Release Candidate` in OctoPrint's Software Update settings for the plugin.

erispe, 2021-03-06:
This is aweome. Thank you very much!

closed 2021-03-07 by jneilliii