jneilliii/OctoPrint-PrusaSlicerThumbnails #69: Thumbnail not being show in list or in popup (I see "broken image link" icon)

james-fry, 2021-03-24:
- PrusaSlicerThumbnails plugin is installed on my octopi.
- Prusaslicer is configured correctly and my gcode has the embedded
- I can see that PNG images have been created in ~/.octoprint/data/prusaslicerthumbnails, and I can see that they are proper working PNGs
- But the UI is not showing the thunbnails. Instead its showing the "broken image link" icon:

[image]

If I open broken image in another window the URL is:
http://octopi//plugin/prusaslicerthumbnails/thumbnail/SKRPLA_makita2ryobi.png?20210324135526

Any ideas whats going wrong?

jneilliii, 2021-03-24:
hmm, what browser? The issue appears to be possibly from the double `//` after the server address.

james-fry, 2021-03-24:
Its Chrome.
And your spotting thedouble // was absolutely on the mark!
Sometimes it takes a 2nd set of eyes :)
I had the // in my bookmark for my octoprint instance.

Bizarrely though octoprint works fine with the // for absolutely everything else - been using it like that for (literally) years...

Anyway all sorted now. Thanks for the help and for the plugin - its really great.

closed 2021-03-24 by jneilliii