jneilliii/OctoPrint-PrusaSlicerThumbnails #70: [Feature Request] - Scan files with button or integrate with Refresh List

FredrikAir, 2021-04-11:
Hey, 
I transfer my files to the uploads folder via SAMBA, so the thumbnails are not generated unless I go into the plugin itself and scan the files.

Is it possible to integrate this function into the "Refresh Files List" or add a new Scan-Files-button to the files list menu?
[image]

jneilliii, 2021-04-11:
have you ever thought about using samba share for the watched folder instead of the uploads folder? uploads folder shouldn't be copied to directly from what I've seen posted o OctoPrint's issue tracker.

FredrikAir, 2021-04-11:
Ok, i did not know that.
I setup the watched folder just now and uploads there generates the thumbnails - Great, Thanks =)

jneilliii, 2021-04-11:
That's great, I'm glad that worked out for you. With that approach all the plugins that utilize file added events will be able to do what they're supposed to. I'll still keep this request open, as I would like to hook the scan files into the refresh button, but currently there are no event hooks I can use within OctoPrint to do it, so would have to be done in an ugly way.

alexandrkut, 2022-01-20:
Hello!
About "watched" folder instead of the "uploads" folder - 
I have 4 identical printers and it is logical to use a shared folder.
Example - i upload file to watched folder - all printers grab the file.
Then I notice that the file is not correct.
Аfter that I have to go to all printers to delete the file, instead of only deleting it on the shared uploads folder.

The rescan button will be very useful in this case (currently i run api request wicurl on each octoprint).

jneilliii, 2022-01-20:
Can't you just re-upload the file to watched folder and have it distribute again. All I'm saying is that the approach of putting files directly into the uploads folder is not a supported use case when it comes to octoprint and I've seen where foosel has mentioned this herself.

alexandrkut, 2022-01-20:
i can reupload. but when i need just delete file?
i will have to delete it on all printers one by one.
" All I'm saying is that the approach of putting files directly into the uploads folder is not a supported" - ok, i understand =)

jneilliii, 2022-01-20:
Just curious @alexandrkut are you using a shared basedir folder for all your instances or just a shared watched folder?

alexandrkut, 2022-01-20:
I use shared upload folder.

jneilliii, 2022-01-21:
New option added with the above commit and part of version 1.0.1rc3. If you want to test it out and verify you can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted.

alexandrkut, 2022-01-21:
Thank you! 
I will test it tomorrow.

PR #92 (2022-10-15): 1.0.1
* fix issues related to gco file extension instead of gcode extensions, resolves #77
* fixes issues related to reserved characters in URLs, resolves #84
* remove unreachable code
* update scan files to be more efficient
* fix bugs with scan files
* new option for rescanning files when refreshing the file list, resolves #70
* automatically grab the longest encoded thumbnail in each file (likely the higher resolution image), thanks to @iviar.
* added Weedo support
* added Luban support, resolves #91
closed 2022-10-15 by jneilliii