jneilliii/OctoPrint-PrusaSlicerThumbnails #71: Oversized plugins

WimVDK, 2021-04-17:
I'm currently testing this plugin (after being used to "UI Customizer, which for the purpose of this test period i've disabled) and installed the required? themify. In prusaslicer i've set thumbnails to 16x16, 256x256

Themify however more than doubles them to 563.617x563.617 and messes up the gcode viewer by placing the bottom scroll bar on the right of the viewing area. Triggering a scrollbar. Considering i always have my bookmarks open in my browser that's a big problem. Is themify to blame here? Is prusathumbnails? I didn't have that problem with "UI Customizer"

[image]

jneilliii, 2021-04-17:
That's what the scaling options are for in the plugin's settings. Have you adjusted those at all?

jneilliii, 2021-04-17:
Just double-checked, that's actually in my current release candidate version. If you want to switch your release channel in software update settings to that you can get the version that has state panel scaling. Would love to get some feedback.

WimVDK, 2021-04-17:
i only just installed it. That's all. couldn't even find the option to resize as you recommend to do in themify

Selector | CSS_Rule | Value
-- | -- | --
#files > div > div.gcode_files > div.scroll-wrapper | min-height | 800px !important

Only have my printer for about 4 months now so still much to learn.

installed rc6 that's the one?

jneilliii, 2021-04-17:
Yes, 0.1.5rc6 is the current version that has more options for scaling and positioning of the thumbnails.

closed 2021-05-25 by github-actions[bot]