jneilliii/OctoPrint-PrusaSlicerThumbnails #72: Support MKS TFT format gcode thumbnails

jneilliii, 2021-04-21:
Moving conversation from https://github.com/Ultimaker/Cura/pull/8890#issuecomment-824040048

@daniel-kukiela we can move the discussion here for now.

daniel-kukiela, 2021-04-21:
For the function that I will write, what's the input that I should expect? Will it be a full G-code in some form, a part of G-code containing the data? pre-processed G-code in some way, ...?

jneilliii, 2021-04-22:
Can you share a file that has been post processed by your script? My initial thought was I would regex extract the chunk of that file that defines the larger thumbnail and pass that along to this decode function with a full path to where the file should be saved and receive a true/false return on success/failure.

daniel-kukiela, 2021-05-10:
Sorry for the long delay.
Here is an example file: https://www.dropbox.com/s/eapuizzlr35wr26/new3_LS_3DBenchy.gcode?dl=0

jneilliii, 2021-05-10:
Thanks, no worries. With all my OctoPrint plugins I have plenty of stuff to tie up my time. I notice this file has 6 thumbnails in it. Is that standard? What I do with the Prusa format is extract out the last one, which in this case appears to be the same, where the last image is the largest? So I would rip this part out and send to the function.

```
;;gimage:00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

daniel-kukiela, 2021-05-10:
I shared the wrong file. This looks like some test from when I was developing this Cura plugin.
Here's the correct one: https://www.dropbox.com/s/kh1yy28g7h3lhzl/LS_wings_0.33.gcode?dl=0

There are 2 thumbnails - smaller (100x100) starting with `;simage:` and bigger (200x200) starting with `;;gimage:`. There also might be just one of them. So the code should choose whatever is available. If you, for example, pass in all lines starting with `;;gimage` or `;simage:` or `M10086` (they'll always be only on the top of the gcode file), I can parse them, choose the right format (the biggest available) and return whatever you want (it could be a numpy array containing image data, png binary string or a BytesIO object, or anything else.

btw. Here's my Cura plugin: https://github.com/daniel-kukiela/LotmaxxSnapshot

jneilliii, 2021-05-11:
Those can be included as dependencies to the plugin. Basically anything available that can be installed from PyPi that runs on arm architecture can be utilized.

daniel-kukiela, 2021-05-11:
It looks like I know all I need to know then.
Since these are small images, I'll see if I can do that without NumPy efficiently.

daniel-kukiela, 2021-05-11:
Yeah, that's one of the reasons I might not use NumPy. I've been there too (compiling it for older RPis).
Then: https://www.piwheels.org/ ?

jneilliii, 2021-05-11:
Yes, both piwheels and pypi are polled during install of OctoPrint plugins.

daniel-kukiela, 2021-05-11:
Ok.
Anyway, like I said, I will try to not use NumPy anyway.

daniel-kukiela, 2021-05-11:
Here's the whole code:
```py
import regex as re
from PIL import Image
import io

gcode = open('LS_wings_0.33.gcode').read()

pattern = re.compile('(?:;(?:simage|;gimage):).*?M10086 ;[\r\n]', re.DOTALL)
gcode_encoded_images = pattern.findall(gcode)

class Test:

    # Extracts a thumbnail from a gcode and returns png binary string
    def extract_thumbnail(self, gcode_encoded_images):

        # Find the biggest thumbnail
        encoded_image_dimensions, encoded_image = self.find_best_thumbnail(gcode_encoded_images)

        # Not found?
        if encoded_image is None:
            return None  # What to return? Is None ok?

        # Remove M10086 ; and whitespaces
        encoded_image = encoded_image.replace('M10086 ;', '').replace('\n', '').replace('\r', '').replace(' ', '')

        # Get bytes from hex
        encoded_image = bytes(bytearray.fromhex(encoded_image))

        # Load pixel data
        image = Image.frombytes('RGB', encoded_image_dimensions, encoded_image, 'raw', 'BGR;16', 0, 1)

        # Save image as png
        with io.BytesIO() as png_bytes:
            image.save(png_bytes, "PNG")
            png_bytes_string = png_bytes.getvalue()

        return png_bytes_string

    # Finds the biggest thumbnail
    def find_best_thumbnail(self, gcode_encoded_images):

        # Check for gimage
        for image in gcode_encoded_images:
            if image.startswith(';;gimage:'):
                # Return size and trimmed string
                return (200, 200), image[9:]

        # Check for simage
        for image in gcode_encoded_images:
            if image.startswith(';simage:'):
                # Return size and trimmed string
                return (100, 100), image[8:]

        # Image not found
        return None

print(Test().extract_thumbnail(gcode_encoded_images))
```
I wrapped it into a Test class and left my other code pieces so you can test it if you want.
It checks if `gimage` is available and if not checks for `simage`. If none of them are available it returns `None`, otherwise png bytes.

jneilliii, 2021-05-12:
Wow, you made this way easier than I needed, thank you so much @daniel-kukiela. I've been able to shim this into the existing plugin fairly easily, just need to do some more testing but initial results look good based on the gcode you sent over.

[image]

jneilliii, 2021-05-12:
[image]

daniel-kukiela, 2021-05-12:
That's great to hear. I'm happy to contribute to your plugin and OctoPrint in general.
If it needs to be updated in the future, or maybe you need some help with something else too, feel free to mention me :)

jneilliii, 2021-05-12:
Thanks. I just released version [0.1.5rc9](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/releases/tag/0.1.5rc9) that adds this support if you want to give it a go. You would just have to change the plugin to use the `Release Candidate` release channel in OctoPrint's Software Update Settings. Please feel free to review the code and see if you see any other efficiency gains that could be had as you appear to be much more astute to python than myself. I hack my way through it until it works usually, lol.

daniel-kukiela, 2021-05-12:
So... I already have Pi 4s and SD cards and I am planning to install OctoPrint. But I still haven't done this, so I also cannot test this myself. Yet. Now, I should finally start using OctoPrint with my printers :)
I'll announce this on the Lotmaxx user groups on Facebook.
I'll also take a look at the rest of the code in the coming days :)

closed 2021-05-12 by jneilliii