jneilliii/OctoPrint-PrusaSlicerThumbnails #79: Plugin doesn't work/cannot be installed

Chodec80, 2021-05-22:
I figured out few days ago that this plugin doesn't work in Octodash v. 2.2.0 and I was surprised that this plugin missing in my Octoprint. I tried to install via Octoprint, via new installation of the Octodash and also manually via terminal.
Installation via terminal shows me following message:

> Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
>     
> Command "python setup.py egg_info" failed with error code 255 in /tmp/pip-req-build-GAdpuv/
> 

[octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6527292/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6527293/plugin_pluginmanager_console.log)

jneilliii, 2021-05-23:
Yes, so in your octoprint.log error it shows the same error as described in #78. This should be resolved by SSH'ing to the device and running `sudo apt install libopenjp2-7`. You may need to `sudo apt update` and `sudo apt upgrade` as well. Once libopenjp2-7 is installed restarting OctoPrint should allow the plugin to initialize.

Chodec80, 2021-05-23:
omg, based on your advice at Discord I tried to install missing dependency before I wrote this post, but I god still the same error about code 255 as I mentioned above. But fact is, that plugin works despite the "error 255".

So, sorry for this unnecessary Issue, you had right with the missing dependency.

closed 2021-05-23 by jneilliii