jneilliii/OctoPrint-PrusaSlicerThumbnails #80: Plugin not installed

JayCee09, 2021-06-09:
I have the newest version of Octodash (v. 2.2.0) and after the upgrade, I found that Prusa Slicer thumbnails plugin is disappeared from the installed plugin list in Octoprint's settings. I tried reinstalling it from the plugin manager, but the successful installation message showed "Unknown" as plugin name, and checking the installed plugins, it wasn't there. Any help would be much appreciated. Thanks in advance!
[image]

JayCee09, 2021-06-09:
[octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6626210/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6626211/plugin_pluginmanager_console.log)
Here are the requested logs

JayCee09, 2021-06-09:
Thanks for your quick response, I tried the command, but this was the result:

Building dependency tree
Reading state information... Done
Package libjpeg-turbo8 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libjpeg-turbo8' has no installation candidate-bash: Reading: command not found

jneilliii, 2021-06-09:
what about `sudo apt install libjpeg-dev`?

JayCee09, 2021-06-09:
Tells me this:

Reading package lists... Done
Building dependency tree
Reading state information... Done
libjpeg-dev is already the newest version (1:1.5.2-2+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

tryed `sudo apt upgrade` and found two packages to upgrade:

libjavascriptcoregtk-4.0-18/stable 2.32.1-1~deb10u1+rpi1 armhf [upgradable from: 2.30.6-1~deb10u1+rpi1]
libwebkit2gtk-4.0-37/stable 2.32.1-1~deb10u1+rpi1 armhf [upgradable from: 2.30.6-1~deb10u1+rpi1]

upgraded them, rebooted and reinstalled  the plugin, but nothing changed: still no Slicer Thumbnails in the installed plugin list.

jneilliii, 2021-06-09:
try this command for me and report back the response please.

```
~/oprint/bin/pip freeze | grep Pillow
```

JayCee09, 2021-06-09:
Here it is:

DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Pillow==6.2.2

jneilliii, 2021-06-09:
Ah, that might be the underlying issue, python 2 vs 3. I don't have an environment to test that anymore. You could upgrade your instance to python 3 and then upgrade Pillow and I think it would resolve the problem. The upgrade script is fairly painless.

https://github.com/cp2004/Octoprint-Upgrade-To-Py3

jneilliii, 2021-06-09:
The other option would be installing the older version and don't upgrade when prompted. Copy/Paste the URL below into plugin manager > get more > ...from URL and clicking Install.
```
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/releases/download/0.1.4/0.1.4.zip
```

JayCee09, 2021-06-09:
Solved upgrading to Python 3 as you suggested. Thanks a lot for your help and patience!

closed 2021-06-09 by jneilliii

yoda-4, 2023-05-18:
Hallo, i have the same issue, my python Version is already 3.9.2

i tried a lot already install OctoPrint 1.9.0rc6 ect...

pi@octopi:~/OctoPrint/venv/bin $ pip install "https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/master.zip
  Using cached https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/master.zip
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-xbo9_5jy/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-xbo9_5jy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-9w1tg8gh
         cwd: /tmp/pip-req-build-xbo9_5jy/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
WARNING: Discarding https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/master.zip. Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
pi@octopi:~/OctoPrint/venv/bin $

jneilliii, 2023-05-18:
@yoda-4 this is actually a bug with OctoPrint and can be fixed by running OctoPrint 1.9.0rc6 (change release channel to maintenance rcs in software update settings) or by manually installing wheel in the venv. Based on the output above I think that would be `~/OctoPrint/venv/bin/pip install wheel`.

yoda-4, 2023-05-19:
@jneilliii thanks for fast reply, as I mentioned I already running 1.9.0rc6, I have the same issue with octoapp, and al slightly different error with octolapse, with most other plugins the installer is working.

pi@octopi:~/OctoPrint/venv/bin $ pip install "https://github.com/FormerLurker/Octolapse/archive/master.zip"
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/Octolapse/archive/master.zip
  Downloading https://github.com/FormerLurker/Octolapse/archive/master.zip
     \ 1.3 MB 1.9 MB/s 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-ezf7e46j/setup.py", line 57, in <module>
          import octoprint.server
      ModuleNotFoundError: No module named 'octoprint'
      likely tags: v0.4.3
      picking v0.4.3
      got version from expanded keyword {'version': 'v0.4.3', 'full-revisionid': '8d53c91069943b9b4d78498c2fc225b3d6d04cb8', 'dirty': False, 'error': None, 'date': '2023-05-13T12:58:31-0500'}
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

yoda-4, 2023-05-19:
I run OctoPrint on a fresh RaspiOS Installation cause of compatibility issues with hyperpixel 4.0

jneilliii, 2023-05-19:
I'd recommend reaching out on the forum [here](https://community.octoprint.org/t/setuptools-error-while-installing-plugin-octoklipper-on-manual-op-installation/51387?u=jneilliii). The wheel install should have resolved the issue. This is a problem with the underlying octoprint install, not the plugin specifically.