jneilliii/OctoPrint-PrusaSlicerThumbnails #81: I can't install it anymore.

acguardia, 2021-07-16:
pi@octobox:~/klipper$ pip install "https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/refs/tags/1.0.0.zip"
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Collecting https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/refs/tags/1.0.0.zip
  Downloading https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/refs/tags/1.0.0.zip
     | 239 kB 204 kB/s
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-foSBMC/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-foSBMC/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-F6DCoJ
         cwd: /tmp/pip-req-build-foSBMC/
    Complete output (3 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

jneilliii, 2021-07-16:
could you please share your otoprint.log and plugin_pluginmanager_console.log? You can drag them into a comment here.

bjwest, 2021-07-17:
I'm getting the same thing.  Here are my logs:

[octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6833936/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6833937/plugin_pluginmanager_console.log)

I've been using my Raspberry Pi for something else and just got a new one.  I inserted the OctoPi SD card, immediately updated to 1.6.1 and got this error afterwards.  I rebooted and got a notification that PrusaSlicer Thumbnails: 1.0.0 was ready to update and received the error again when I tried to upgrade it.

jneilliii, 2021-07-18:
@bjwest can you share your plugin_softwareupdate_console.log as well please?

bjwest, 2021-07-18:
Sure thing.
[plugin_softwareupdate_console.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6835743/plugin_softwareupdate_console.log)

jneilliii, 2021-07-19:
@njwest, I think I see the issue for you. try copying/pasting this URL directly in Plugin Manager > Get More > ...from URL.

```
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/refs/tags/1.0.0.zip
```

bjwest, 2021-07-19:
Thank you jneilliii, that worked.

nysports, 2021-07-23:
I keep trying to install and get the below followed by "Installing Unknow":

2021-07-23 02:58:19,472   /home/pi/OctoPrint/venv/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp8m0zqfxg/OctoPrint-PrusaSlicerThumbnails-master.zip --no-cache-dir
2021-07-23 02:58:22,914 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-07-23 02:58:22,915 > Processing /tmp/tmp8m0zqfxg/OctoPrint-PrusaSlicerThumbnails-master.zip
2021-07-23 02:58:22,916 > Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from Slicer-Thumbnails==1.0.0) (1.6.1)
2021-07-23 02:58:22,917 > Requirement already satisfied: Pillow in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from Slicer-Thumbnails==1.0.0) (8.2.0)
2021-07-23 02:58:22,917 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (7.1.2)
2021-07-23 02:58:22,917 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2.0)
2021-07-23 02:58:22,917 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.0.0)
2021-07-23 02:58:22,917 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.5.0)
2021-07-23 02:58:22,917 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2.11.2)
2021-07-23 02:58:22,917 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2021.2.23)
2021-07-23 02:58:22,918 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.2.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2021.5.31)
2021-07-23 02:58:22,918 > Requirement already satisfied: OctoPrint-PiSupport>=2021.3.26 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2021.6.14)
2021-07-23 02:58:22,918 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (5.4.1)
2021-07-23 02:58:22,918 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.4)
2021-07-23 02:58:22,918 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.1.1)
2021-07-23 02:58:22,918 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.6.0)
2021-07-23 02:58:22,919 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.0.7)
2021-07-23 02:58:22,919 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.1.2)
2021-07-23 02:58:22,919 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.18.2)
2021-07-23 02:58:22,919 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.1.0)
2021-07-23 02:58:22,919 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (3.1.1)
2021-07-23 02:58:22,919 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.1.1)
2021-07-23 02:58:22,919 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.8.0)
2021-07-23 02:58:22,919 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (0.10.9)
2021-07-23 02:58:22,920 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.7.0)
2021-07-23 02:58:22,920 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (5.8.0)
2021-07-23 02:58:22,920 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (1.2.0)
2021-07-23 02:58:22,920 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (3.5)
2021-07-23 02:58:22,920 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->Slicer-Thumbnails==1.0.0) (2020.11.13)
[... cut ...]
2021-07-23 02:58:22,924 > Attempting uninstall: Slicer-Thumbnails
2021-07-23 02:58:22,924 > Found existing installation: Slicer-Thumbnails 1.0.0
2021-07-23 02:58:22,924 > Uninstalling Slicer-Thumbnails-1.0.0:
2021-07-23 02:58:22,924 > Successfully uninstalled Slicer-Thumbnails-1.0.0
2021-07-23 02:58:22,924 > Running setup.py install for Slicer-Thumbnails: started
2021-07-23 02:58:23,425 > Running setup.py install for Slicer-Thumbnails: finished with status 'done'
2021-07-23 02:58:23,426 > Successfully installed Slicer-Thumbnails-1.0.0

jneilliii, 2021-07-23:
can you share your octoprint.log @nysports, because the install states successful. I think this issue is dependency related.

njwest, 2021-07-24:
> @njwest, I think I see the issue for you. try copying/pasting this URL directly in Plugin Manager > Get More > ...from URL.
> 
> ```
> https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/refs/tags/1.0.0.zip
> ```

wrong username, i'm njwest, you're looking for bjwest

bjwest, 2021-07-24:
@njwest 
> wrong username, i'm njwest, you're looking for bjwest

No worries.  I didn't even notice the wrong username and followed the instructions anyway.  Worked like a charm.

nysports, 2021-08-01:
> can you share your octoprint.log @nysports, because the install states successful. I think this issue is dependency related.

Sorry for the delayed response. I was away.  Log attached.
[octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6913330/octoprint.log)

jneilliii, 2021-08-16:
> ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/issues/78#issue-898897452

closed 2021-08-16 by jneilliii

nysports, 2021-08-25:
Hey @jneilliii wondering if you saw this. Appreciate your help. 

> > can you share your octoprint.log @nysports, because the install states successful. I think this issue is dependency related.
> 
> Sorry for the delayed response. I was away. Log attached.
> [octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/6913330/octoprint.log)

jneilliii, 2021-08-25:
I posted a link to another issue with the fix for your problem.

nysports, 2021-08-28:
Hey, success! Really nice of you to help. I appreciate it.