jneilliii/OctoPrint-PrusaSlicerThumbnails #84: Reserved characters in thumbnail filenames are not correctly encoded (leading to 404 error)

manolitto, 2022-01-06:
The src attribute of the generated img element is not correctly encoded.
All reserved characters must be percent encoded: see "Reserved characters" in https://en.wikipedia.org/wiki/Percent-encoding

### Example:

`<img id="prusaslicer_state_thumbnail" class="pull-left" src="plugin/prusaslicerthumbnails/thumbnail/Base#magnetic,openlock_0.15mm_PLA_MK3S_5h17m.png?20220106190630" width="100%">`

--> gives a 404 not found error

... whereas

`<img id="prusaslicer_state_thumbnail" class="pull-left" src="plugin/prusaslicerthumbnails/thumbnail/Base%23magnetic,openlock_0.15mm_PLA_MK3S_5h17m.png?20220106190630" width="100%">`

--> correctly displays the image

jneilliii, 2022-01-06:
Thanks for this. I thought I had accounted for that already, but will definitely take a look.

SturmB, 2022-07-20:
Good evening. I'm afraid this issue is not completely fixed. I have tried both the latest stable (1.0.0) and RC (1.0.1rc5) and am witnessing the same issue with both. I have detailed the issue on [this Reddit thread](https://www.reddit.com/r/octoprint/comments/vs6bqs/no_more_thumbnails/).

In short, whenever the gcode filename contains a special character, such as '#' or '%', the thumbnail renderer breaks and displays the 'broken image' icon instead of the image. (Seems to return a 400 error code.) I have also confirmed that renaming an existing gcode file to remove the special characters fixes the issue.

Some info that may be of use for you:

* OctoPrint 1.8.1
* OctoPi 0.18.0
* Prusa i3 MK3S+ running firmware 3.10.1-4697
* Chrome 103.0.5060.114
* [octoprint-systeminfo-20220719223302.zip](https://github.com/OctoPrint/OctoPrint/files/9146039/octoprint-systeminfo-20220719223302.zip)
* [gcode_files.zip](https://github.com/OctoPrint/OctoPrint/files/9144933/gcode_files.zip)
* [octopi.solamnia.xyz-1658284542429.log](https://github.com/OctoPrint/OctoPrint/files/9146043/octopi.solamnia.xyz-1658284542429.log)
* Screenshot comparison with/without special character in filename:
[image]
[image]

jneilliii, 2022-07-20:
Did you reupload your file after updating to the latest rc version?

SturmB, 2022-07-21:
Just tried it. Yes, uploading a gcode with a special character in its filename while the v1.0.1rc5 Slicer Thumbnail plugin is installed _does_ show the thumbnail. Yay!

Existing files on the OctoPi still need to have their filenames changed, however. Or deleted and re-uploaded.

jneilliii, 2022-07-21:
Thanks for the confirmation @SturmB. I'm pretty sure the scan files option in settings wouldn't pick these up either because when it checks the metadata the filename matches what's there and it's purely a web encoding issue.

SturmB, 2022-07-21:
I just tried the "Scan Files" option, just for the heck of it. You are correct that thumbnails for gcode files that were uploaded while using v1.0.0 of the plugin are still not recognized.

I look forward to RC5 making it into stable. In the mean time, I'll probably keep using RC5 as I don't see any bugs with it so far. I might also try to refrain from adding special characters to my filenames anyway, just to prevent any other possible problems in the future, perhaps with other plugins.

jneilliii, 2022-07-21:
you can actually disable the newly added feature in OctoPrint if you are just trying to avoid the special characters in general. I'll try to add some logic for the update install to handle the migration a little better between versions. might be able to do that tonight and release a rc6 version to validate it works properly prior to stable release. you'll find most my plugins have rc versions newer than stable release and are more or less fully functional. I typically put those out there for people to validate changes related to reported issues prior to stable release.

PR #92 (2022-10-15): 1.0.1
* fix issues related to gco file extension instead of gcode extensions, resolves #77
* fixes issues related to reserved characters in URLs, resolves #84
* remove unreachable code
* update scan files to be more efficient
* fix bugs with scan files
* new option for rescanning files when refreshing the file list, resolves #70
* automatically grab the longest encoded thumbnail in each file (likely the higher resolution image), thanks to @iviar.
* added Weedo support
* added Luban support, resolves #91
closed 2022-10-15 by jneilliii