jneilliii/OctoPrint-PrusaSlicerThumbnails #89: Extracted Thumbnail in Source GCode Path?

jlong23, 2022-02-17:
Since the plugin modifies the ".metadata.json" with the paths of the thumbnails, would it be possible to store the extracted thumbs with the same path as of the GCode?   

Reason is that if you have a NFS Share of all your GCode with multiple Printers pointing to it, it's now shared and the Thumbs match the Shared ".metadata.json" paths as well.

bolsoncerrado, 2022-12-01:
So to try and further troubleshoot this.... 

I just uploaded 2 gcode files in 2 sepparate Pi's, the one in the top left as you can see and a Pi located to its right.

As you can see in the right half of the screenshot, the thumbnails were stored correctly on the NAS folder.

[image]

However NONE of the uploaded files show up with the thumbnail icon available, nor  no thumbnail is loaded after upload finishes  :?

However I'm pretty sure if I reupload the same files again, they will show the thumbs fine :????

bolsoncerrado, 2022-12-01:
> However I'm pretty sure if I reupload the same files again, they will show the thumbs fine :????

And this just happened! I reuploaded the same files on the same Pis and now the thumbs showed fine.

I sliced and uploaded a totally different gcode on a totally different Pi and same scenario: first, no thumbnail. Reupload+Overwrite=thumbnail

:??????

jneilliii, 2022-12-01:
The only thing I can think of in this scenario is a possible caching problem. When you upload once to one pi does the thumbnail show up in that instances' interface? What happens when you press the refresh file list button? Any errors in browser's console log?

bolsoncerrado, 2022-12-02:
That's what in trying to explain. First upload, no errors, no thumbnail shows (but thumbnail appears on the Nas drive). Second upload, thumbnail shows fine.

bolsoncerrado, 2022-12-02:
Forgot to add, pressing refresh does not solve the problem.

I may add this may be connected with the fact some Pi's don't show the thumbnail logo of files uploaded on other Pi's even if now the folder for the thumbnails is shared...

SO I guess I'm still missing a symlink or something like that somewhere...

bolsoncerrado, 2022-12-04:
Hey @jneilliii mind helping me understand the chain of events after upload starts on the Pi?

I mean, it's clear somehow the files ARE uploaded and no matter what, the THUMBNAILS are extracted too, even if they don't show up on file refresh list.

Where is the path to the thumbnails stored and why it would work after several uploads? I mean, if symlink finds them anyway after a few reuploads, what else could be failing?

Thanks!

jneilliii, 2022-12-04:
the workflow is at such. 

1. file added via OctoPrint's UI and plugin starts the work [here](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/9765fe44401eb4989a7ef87750c11c20832fb699/octoprint_prusaslicerthumbnails/__init__.py#L204)
2. thumbnail gets extracted [here](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/9765fe44401eb4989a7ef87750c11c20832fb699/octoprint_prusaslicerthumbnails/__init__.py#L214).
3. if extraction was successful and file exists afterwards then the virutal path to the thumbnail is added to `.metadata.json` [here](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/9765fe44401eb4989a7ef87750c11c20832fb699/octoprint_prusaslicerthumbnails/__init__.py#L218).

At that point the URL for the thumbnail is in the metadata OctoPrint uses to display files. 

This is one of the biggest reasons OctoPrint core maintainer (foosel/Gina) doesn't recommend using a shared uploads folder. I think what is happening is all the files are competing to analyze the file, write the metadata, etc. at the same time. You'd be better off using something that syncs the pi's uploads folder like GitFiles, but not the one registered on the repo, use this fork: https://github.com/tideline3d/OctoPrint-GitFiles

jneilliii, 2022-12-04:
Commit above adds the option in settings to `Extract images into uploads folder.` and is part of 1.0.2rc1. You can change the release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted to test it out and see if it resolves the issue or not. I'm not very hopeful base on the above comments that it will work as symlinks are still involved in this scenario.

jneilliii, 2022-12-04:
Only issue I've seen so far is if you have Custom Backgrounds plugin installed, then extracted images are displayed in the file list. That will have to be corrected in that plugin though.

jneilliii, 2023-03-11:
included in the above release 1.0.2

closed 2023-03-11 by jneilliii