jneilliii/OctoPrint-PrusaSlicerThumbnails #91: [Feature Request] Support Luban thumbnails

Rocket78, 2022-10-04:
Luban is the slicer from Snapmaker and is also embedding thumbnails in gcode files to show them on the touch display. It is preceeded by a ";thumbnail:" comment following a whitespace and a html like formatting of the image data. Should be very easy to implement. I attached a sample:
[3DBenchy_1664841208541.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/9701798/3DBenchy_1664841208541.zip)

I also use the Snapmaker 2.0 Cura plugin (https://github.com/macdylan/Snapmaker2Plugin) which could send the  gcode via wifi directly after slicing which also adds the needed info for the touch display including the thumbnail. But as I am using Octoprint most of the time it would be nice to see these thumbnails.

jneilliii, 2022-10-04:
oh yeah, this would be fairly easy to implement in the plugin. the format is the exact same (base64 encoded png file) but on one line compared to how PrusaSlicer/Cura Thumbnails post processor does it.

jneilliii, 2022-10-04:
Added with the above commit and available as version 1.0.1rc6. You can change the release channel for the plugin in OctoPrint's Software Update settings to `Release Candidate` and update when prompted to get the changes. Unfortunately the size of the image is pretty tiny.

[image]

Rocket78, 2022-10-04:
Thanks for the fast reply and implementation. The size of the example is 720x480px when I paste the Base64 here https://base64.guru/converter/decode/image which is fairly big imho. Are we talking about the same? The Snapmaker Cura plugin produces 150x150px thumbnails which is also enough for me.

jneilliii, 2022-10-05:
It could have been the settings I had in my dev environment. I honestly didn't look at the actual extracted png file.

PR #92 (2022-10-15): 1.0.1
* fix issues related to gco file extension instead of gcode extensions, resolves #77
* fixes issues related to reserved characters in URLs, resolves #84
* remove unreachable code
* update scan files to be more efficient
* fix bugs with scan files
* new option for rescanning files when refreshing the file list, resolves #70
* automatically grab the longest encoded thumbnail in each file (likely the higher resolution image), thanks to @iviar.
* added Weedo support
* added Luban support, resolves #91
closed 2022-10-15 by jneilliii