jneilliii/OctoPrint-PrusaSlicerThumbnails #93: Version 1.01 tries to repeatly update.

DaBigPenguin, 2022-10-18:
Logging in to octoprint I continually get a message to update Slicer Thumbnails to 1.0.1 the update runs and appears successful and the notification to update is repeated. I manually updated and it still wants to notify about the update as if it is not updated.

/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/1.0.1.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps                                              Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/1.0.1.zip
  Downloading https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/1.0.1.zip (300 kB)
     |████████████████████████████████| 300 kB 800 kB/s
Building wheels for collected packages: Slicer-Thumbnails
  Building wheel for Slicer-Thumbnails (setup.py) ... done
  Created wheel for Slicer-Thumbnails: filename=Slicer_Thumbnails-1.0.1-py3-none-any.whl size=10605 sha256=7571c6db5fe778dd3b82c0d38285771ab039aefaac76cd66e37426be2b57bf99
  Stored in directory: /tmp/pip-ephem-wheel-cache-fuiuledo/wheels/c9/29/d3/f8bea897f8bdbcfe82618570699d14b472798a893ccecc1c10
Successfully built Slicer-Thumbnails
Installing collected packages: Slicer-Thumbnails
Successfully installed Slicer-Thumbnails-1.0.1

jneilliii, 2022-10-18:
Can't seem to reproduce this issue on my end. Try removing the plugin (don't clear settings) and reinstall from plugin manager.

DaBigPenguin, 2022-10-18:
Tried that too and it does the same thing.

jneilliii, 2022-10-18:
I'm at a loss as to why this would be happening for you. I have two instances and have upgraded both and this isn't happening for me. Can you share your octoprint.log and plugin_pluginmanager_console.log from the last attempt? If you go into software update settings and change the release channel to `Release Candidate` for the plugin does handle it any different?

DaBigPenguin, 2022-10-18:
[octoprint-logs(4).zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/9815176/octoprint-logs.4.zip)

jneilliii, 2022-10-18:
this is very strange. I only see one potential issue:

```
2022-10-15 15:03:51,644 ! ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: 'top_level.txt'
2022-10-15 15:03:51,645 ! Consider using the `--user` option or check the permissions.
```
around the time of the uninstall I suspect. I do not know that is, but see if checking the box in plugin manager's settings (wrench icon) at the bottom for enabling `--user` will allow for a full uninstall of the previous version.

DaBigPenguin, 2022-10-18:
Looks like there is a bug there. Uninstalling reports an error, looks like a typo -licer is missing an "s". 

Uninstalling plugin "Slicer Thumbnails"/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check uninstall --yes -licer-ThumbnailsUsage:/home/pi/oprint/bin/python3 -m pip uninstall [options] <package> .../home/pi/oprint/bin/python3 -m pip uninstall [options] -r <requirements file> ...no such option: -l

jneilliii, 2022-10-18:
Weird, what does '/home/pi/oprint/bin/python3 -m pip freeze` list after uninstalling?

DaBigPenguin, 2022-10-18:
Looks like uninstalling via the plugin manager and restarting does not actually uninstall the plugin.

pi@octoprint:~ $ /home/pi/oprint/bin/python3 -m pip freeze
-ctoPrint-PrintTimeGenius @ https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip
-ctoPrint-SpoolManager @ https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
-licer-Thumbnails @ https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/master.zip
anyio==3.6.1
Arc-Welder @ file:///tmp/tmpdabba96u/ArcWelderPlugin-master.zip
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asyncclick==8.1.3.2
awesome-slugify==1.6.5
Babel==2.10.3
backoff==2.1.2
backports.csv==1.0.7
Backup-Scheduler @ file:///tmp/tmp394w623t/OctoPrint-BackupScheduler-master.zip
blinker==1.5
bson==0.5.10
cachelib==0.2.0
cachetools==5.2.0
certifi==2022.9.14
cffi==1.15.1
Change-Filament @ file:///tmp/tmpo66qa15p/Change_Filament-master.zip
chardet==4.0.0
charset-normalizer==2.1.1
click==8.1.3
colorlog==6.7.0
colorzero==2.0
commonmark==0.9.1
Creality-2x-temperature-reporting-fix @ file:///tmp/tmpiq8f2ox8/OctoPrint-Creality2xTemperatureReportingFix-master.zip
cryptography==37.0.4
Deprecated==1.2.13
DisplayLayerProgress @ file:///tmp/tmppnvd9zy6/master.zip
distlib==0.3.6
docker==2.7.0
docker-pycreds==0.4.0
emoji==1.7.0
feedparser==6.0.10
file-read-backwards==2.0.0
filelock==3.8.0
filetype==1.1.0
Flask==2.1.3
Flask-Assets==2.0
Flask-Babel==2.0.0
Flask-Limiter==2.6.2
Flask-Login==0.5.0
frozendict==2.3.4
future==0.18.2
google-api-core==2.8.2
google-api-python-client==2.57.0
google-auth==2.11.0
google-auth-httplib2==0.1.0
googleapis-common-protos==1.56.4
httplib2==0.20.4
idna==3.4
ifaddr==0.2.0
importlib-metadata==4.12.0
intelhex==2.3.0
itsdangerous==2.1.2
Jinja2==3.1.2
limits==2.7.0
Markdown==3.4.1
MarkupSafe==2.1.1
Marlin-EEPROM-editor @ file:///tmp/tmpyhrl83c7/release.zip
mock==4.0.3
netaddr==0.8.0
netifaces==0.11.0
oauth2client==4.1.3
OctoDash-Companion @ https://github.com/jneilliii/OctoPrint-OctoDashCompanion/archive/master.zip
octoflatbuffers==2.0.3
Octolapse @ file:///tmp/tmpyr66upt2/Octolapse-master.zip
OctoPrint @ https://github.com/OctoPrint/OctoPrint/archive/1.8.6.zip
OctoPrint-BLTouch @ file:///tmp/tmposqq5qb7/OctoPrint-BLTouch-master.zip
OctoPrint-CalibrationTools @ file:///tmp/tmpqozqg9nl/OctoPrint-CalibrationTools-main.zip
OctoPrint-CameraSettings @ file:///tmp/tmpmr4pp36f/OctoPrint-CameraSettings-main.zip
OctoPrint-CrealityTemperature @ file:///tmp/tmphsimhmli/master.zip
OctoPrint-ExtraFileInfo @ file:///tmp/tmpmz78_nms/OctoPrint-ExtraFileInfo-master.zip
OctoPrint-FileCheck==2021.2.23
OctoPrint-FirmwareCheck==2021.10.11
OctoPrint-FirmwareUpdater @ file:///tmp/tmp0oy36pm2/OctoPrint-FirmwareUpdater-master.zip
OctoPrint-GcodeEditor @ file:///tmp/tmpet37xhlj/OctoPrint-GcodeEditor-master.zip
OctoPrint-HeaterTimeout @ file:///tmp/tmpsahylmg7/OctoPrint-HeaterTimeout-master.zip
OctoPrint-ipOnConnect @ file:///tmp/tmpc5xntzeu/OctoPrint-ipOnConnect-master.zip
OctoPrint-PiSupport==2022.6.13
OctoPrint-PrintTimeGenius @ https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip
OctoPrint-Resource-Monitor @ file:///tmp/tmpcjqpjgyf/OctoPrint-Resource-Monitor-master.zip
OctoPrint-SimpleEmergencyStop @ file:///tmp/tmp8h6vm2ds/OctoPrint-SimpleEmergencyStop-master.zip
OctoPrint-SlicerSettingsParser @ https://github.com/Rob4226/OctoPrint-SlicerSettingsParser/archive/master.zip
OctoPrint-Tasmota @ https://github.com/jneilliii/OctoPrint-Tasmota/archive/1.1.3.zip
OctoPrint-UltimakerFormatPackage @ file:///tmp/tmpi3y3lxit/OctoPrint-UltimakerFormatPackage-master.zip
packaging==21.3
passlib==1.7.4
pathtools==0.1.2
pathvalidate==2.5.2
peewee==3.15.1
picamerab==1.13b1
Pillow==9.2.0
pkg-resources==0.0.0
pkginfo==1.8.3
platformdirs==2.5.2
Preheat @ https://github.com/marian42/octoprint-preheat/archive/master.zip
protobuf==4.21.5
psutil==5.9.2
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
pydantic==1.9.2
PyDrive2==1.14.0
pyduinocli==0.23.0
Pygments==2.13.0
pylru==1.2.1
pyOpenSSL==22.0.0
pyparsing==3.0.9
pyserial==3.5
python-dateutil==2.8.2
python-kasa==0.5.0
pytz==2022.2.1
PyYAML==5.4.1
qrcode==7.3.1
regex==2022.9.13
requests==2.28.1
rich==12.5.1
rsa==4.0
sarge==0.1.6
semantic-version==2.10.0
sentry-sdk==1.5.12
sgmllib3k==1.0.0
six==1.15.0
Slicer-Thumbnails @ https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/archive/1.0.1.zip
sniffio==1.2.0
SQLAlchemy==1.1.18
tornado==6.2
treelib==1.6.1
typing-extensions==4.3.0
UI-Customizer @ file:///tmp/tmprcvlkfzp/OctoPrint-UICustomizer-main.zip
Unidecode==1.3.4
uptime==3.0.1
uritemplate==4.1.1
uritools==2.1.1
urllib3==1.26.12
virtualenv==20.16.5
watchdog==1.0.2
webassets==2.0
websocket-client==1.4.1
Werkzeug==2.0.3
WiFi-Status @ file:///tmp/tmp_mzjir7y/OctoPrint-WiFiStatus-main.zip
wrapt==1.13.3
Z-Probe-Offset-Control @ file:///tmp/tmpbeoaf5eo/octoprint_z_probe_offset-latest.zip
zeroconf==0.33.4
zipp==3.8.1
zipstream-ng==1.3.4

jneilliii, 2022-10-19:
Yeah, looks like a few there that are oddly named. Try `pip uninstall  -licer-Thumbnails` and see if that could get rid of the top one.

DaBigPenguin, 2022-10-19:
Takes it as a unknown switch. 

pi@octoprint:~ $ /home/pi/oprint/bin/python3 -m pip uninstall -licer-Thumbnails

Usage:
  /home/pi/oprint/bin/python3 -m pip uninstall [options] <package> ...
  /home/pi/oprint/bin/python3 -m pip uninstall [options] -r <requirements file> ...

no such option: -l
pi@octoprint:~ $ /home/pi/oprint/bin/python3 -m pip uninstall "-licer-Thumbnails"

Usage:
  /home/pi/oprint/bin/python3 -m pip uninstall [options] <package> ...
  /home/pi/oprint/bin/python3 -m pip uninstall [options] -r <requirements file> ...

no such option: -l

DaBigPenguin, 2022-10-19:
I did some digging and in ~/oprint/lib/python3.7/site-packages renaming the dir to what it should be appears to have allowed the uninstall to at least partially work with one of the other oddly named modules. It also is a tilde and not a hyphen as I first thought. Now how they got in that state I am not sure.

jneilliii, 2022-10-19:
You could do the same for the slicer thumbnails plugin maybe. I would just attempt deleting both folders.

jneilliii, 2022-10-19:
I just verified, deleting the `/home/pi/oprint/lib/python3.7/site-packages/Slicer_Thumbnails-1.0.1.dist-info` folder in my instance and running `sudo service octoprint restart` completely gets rid of the plugin and I was able to install from Plugin Manager.

closed 2022-11-11 by github-actions[bot]