jneilliii/OctoPrint-PrusaSlicerThumbnails #95: Randomly working on MacOS BigSUR

bolsoncerrado, 2022-11-30:
Howdy,

I'm posting the issue here as well because I think it's NOT related to the OS  but any insights are welcome!

I'm trying simplify3D thumbnails in our raspberry pi-driven Prusa MK3S farm but I find it odd: the thumbnail generation works fine up to where I know (I can open both .gcode files and both have the thumbnail code inside), but once I upload them to the SHARED uploads folder all raspberries look at, some thumbnails show up, some don't and the ones that show up on one raspberry show as broken on another :?

Any clues what could be going on here?

[image]

In this screenshot you can see top left most "octoprint" had ROBOCOP AUTO9 uploaded and thumbnail displaying fine.
To its right you can see the call to the thumbnail from the octoprint file icon and it won't load the thumbnail.
And on the lower row I uploaded COLT MKIV which has thumbnails on the gcode but its not detected by octoprint.

Yes, ALL instances have Slicer Thumbnails plugin enabled and I've already tried restarting/rebooting the PIs... 

Any clues?

Thanks!

jneilliii, 2022-11-30:
> SHARED uploads folder

this is the reason why. the plugin isn't really designed well for this scenario as the thumbnails are extracted into individual png files stored in the instance's basedir location inside data/prusaslicerthumbnails/. because of this the file may exist on one instance but not other instances. if all instances are running on a single machine you could symlink that data folder to all instances and that would work around the issue. 

a similar request #89 was made to store these extracted files directly into the uploads folder, which would also resolve this issue for you I think. since it's the same, closing this as duplicate.

closed 2022-11-30 by jneilliii

bolsoncerrado, 2022-11-30:
>  if all instances are running on a single machine you could symlink that data folder to all instances and that would work around the issue.

Each printer has its own raspi, each raspi uses the same cloned octopi image, plugins and settings, same uploaded folder on a NAS server...

Can a symlink still be used in this scenario?

Thanks for the insight.

bolsoncerrado, 2022-11-30:
OH btw just 1 more thing now that we are at it :P 

How do I "move" the captured frame a bit lower? I've read the docs about which lines to edit on the bash script but I cant understand the logic behind it.

So say I have this thumbnail:

[image]

...but I want to take the "snapshot" somewhat lower so all the bed gets in the pic.... 

How would I edit the height line on the bash script?

Thanks!

jneilliii, 2022-11-30:
sorry, can't help you there as I don't utilize Simplify3D or know anything about those scripts that capture the image.

bolsoncerrado, 2022-12-01:
Gotcha Many Thanks.